VR-25/acc #1: Not working on Asus Nexus 7

hapklaar, 2019-05-20:
Unfortunately acc isn't working for the Nexus 7 (2013 flo model). I've uploaded /sbin/_acc/acc-power_supply-flo.log to your dropbox and hope there is a way to get this working for this older tablet :)

Please let me know if you need anything else from me to help!

VR-25, 2019-06-09:
Hi,
I'm yet to find a suitable charging control file for your device.
I've been working with another flo user to find one.
We even tried ElementalX kernel - no use.
I'll keep digging...
Keep this issue open.

hapklaar, 2019-06-11:
Hi, 
Thanks for letting me know and hope you find something!

ryperx, 2020-11-12:
@VR-25 Do you found any solution? 
I am also looking for a solution for the Nexus 7 2012 and it would be great! I want to let my Nexus 7 plugged in all the time and this would help greatly!

hapklaar, 2020-11-12:
fyi In the mean time I solved it by measuring the battery level and switching the outlet based on battery level. Over 80% turn off outlet,, below 40% turn on outlet. All done with Home Assistant.

ryperx, 2020-11-12:
@hapklaar 
thanks for you response. This is an interesting solution, never had thought oif that.
Would prefer a possibility with ACC but i will consider this.

Thanks!

VR-25, 2020-11-13:
ACC now has a feature called _current-based charging control_.
I was about to ask you to upload logs from both models.
The log archive is generated by `acc -le`.
I'll see if there are charging current control files.

ryperx, 2020-11-13:
[acc-logs-grouper.zip](https://github.com/VR-25/acc/files/5535209/acc-logs-grouper.zip)

Attached the log file, i hope it helps to get the feature enabled :)

ryperx, 2020-11-24:
@VR-25 
Any Update on this or how i could help?
It would be a great help if acc works on the nexus

ryperx, 2020-11-29:
I would like to create another log files, but the problem is after using acc -l -e  no log file zip is created.
Any idea what is causing this? I have acc installed with magisk and it is on latest version

VR-25, 2021-07-21:
Still, there are no current/voltage control files for _grouper_ nor _flo_ either.
I found a potential charging switch for _grouper_: `/sys/devices/platform/tegra-i2c.4/i2c-4/4-0055/battery_smbus 1 0`.
See if writing `0` to that file stops charging.
Note, this may fail and cause a system freeze, followed by a reboot - nothing to worry about, though.

> I would like to create another log files, but the problem is after using acc -l -e no log file zip is created.
> Any idea what is causing this? I have acc installed with magisk and it is on latest version
> 
> Edit:
> The latest rar file that was created is broken but i got some .log files. I attached them
> 
> [logs.zip](https://github.com/VR-25/acc/files/5612626/logs.zip)

closed 2021-07-22 by VR-25

lelemka0, 2022-09-08:
acc until now still can't stop charging on nexus7 (2013). I use lineage-19.1 with ElementalX kernel.
https://forum.xda-developers.com/t/disable-charging-decrease-charging-voltage-on-nexus-7-2013-solved.3682984/post-74970008
I found this thread and tried `echo 0 > /sys/devices/i2c-0/0-0055/battery_smbus`, but charging still won't stop.
```
#cat /sys/devices/i2c-0/0-0055/battery_smbus
1
```
[log.zip](https://github.com/VR-25/acc/files/9531098/log.zip)