VR-25/acc #2: Not working on Vernee Apollo Lite

curtisy1, 2019-06-13:
Hey, I'm using a Vernee Apollo Lite (exotic, I know) with MadOS as custom ROM. When starting the Daemon, I get the following output:
```
echo /sys/devices/soc.0/qpnp-linear-charger-8/power_supply/battery_qpnp/charging_enabled 1 0
awk '{print $1}
'[-f /sys/devices/soc.0/qpnp-linear-charger-8/power_supply/battery_qpnp/charging_enabled]'
IFS= read -r file
false
exxit 1 '(!) Unsupported device'
typeset 'exitCode=1'
set +euxo pipefail
(!) Unsupported device
***EXIT 1***
``` 

If I can help to fix this somehow, please let me know!

VR-25, 2019-06-14:
Hi,
This means no suitable charging control file was found on your system.
Send me `/sbin/.acc/acc-power_supply-*.log`.

curtisy1, 2019-06-24:
Sorry for the delay. Here's the output log
[acc-logs-apollo_lite.zip](https://github.com/VR-25/acc/files/3322416/acc-logs-apollo_lite.zip)

VR-25, 2020-01-22:
MediaTek devices support is finally coming.
Stay tuned for the next release!

VR-25, 2020-02-25:
MediaTek support is live.

closed 2020-02-25 by VR-25