VR-25/acc #4: Full charge / Frustration override

eugenesan, 2019-06-16:
This is more of a feature request than an issue.

Issue A:
---
You have an unusually long day ahead of you. You put your device on charge but then remember it won't charge fully. Of course, you could just change the setting before and after but that's a hassle!

Issue B:
---
You've asked a person unfamiliar with your charging customizations, to put the device on charge. They plug it in - nothing happens, they try again and again... They are frustrated and your charging port is possibly damaged!

Solutions:
---
Once the daemon detects plugin event, it monitors for a second one for 5-10 seconds and if the second event detected, it temporarily (until it charges or another plugin event) changes the charging limit to 100%.

Action:
---
A. If you want a full charge, just plugin in the charger twice in 5-10 seconds.
B. If someone unfamiliar plugs in your device and fails to see/hear charging indication they will retry and succeed assuming that the first attempt was a fluke.

VR-25, 2019-06-17:
Interesting

This requires a dedicated charger plugin event listener that would communicate with accd via flag/lock files.
A major issue would be the reliability of such events. I'll think more about that.

Currently, you can run `acc -f` to charge once without limitations.
With Tasker, double-pressing the power button to open camera can be reprogrammed to run that command.

[Screenshot_20190617-074344](https://user-images.githubusercontent.com/28943176/59584436-f0915180-90d5-11e9-80b3-7d2b5ed3f0ad.png)

VR-25, 2019-07-08:
I may have found a way to reliably detect plug in events, and run `acc --full` accordingly - to charge fully once, then restart the daemon with regular settings.

Here's a little homework for you - to test that possibility:

What's the output of `su -c acpi -a`?...

1. When unplugged
2. When plugged in, charging (`acc -e`)
2. When plugged in, NOT charging (`acc -d`)

elem089, 2019-07-08:
for me (OP6) "su - acpi -a" delivers 3 lines for "Adapter 0", "Adapter 1" and "Adapter 2". Only visible change is in "Adapter 0"
1. Adapter 0: off-line
2. Adapter 0: on-line
3. Adapter 0: off-line

VR-25, 2019-07-08:
@elem089, acpi won't be useful for OP6.

Now, what are the outputs of `su -c dumpsys battery` and `su -c dumpsys batteryproperties`, respectively, when unplugged, not charging (acc -d)?

Thanks

elem089, 2019-07-08:
ok, didn't know that, good that I mentioned the device ;)
here we go: "dumpsys batteryproperties" is always empty so the results are for "dumpsys battery":

1. unplugged:
`:/ $ su -c dumpsys battery
Current Battery Service state:
AC powered: false
USB powered: false
Wireless powered: false
Max charging current: 0
Max charging voltage: 0
Charge counter: 2425142
status: 3
health: 2
present: true
level: 84
scale: 100
voltage: 4146
temperature: 269
technology: Li-ion
fastChargeStatus: false
`
2. plugged, charging:
`:/ $ su -c acc -e
(i) Charging enabled
:/ $ su -c dumpsys battery
Current Battery Service state:
AC powered: false
USB powered: true
Wireless powered: false
Max charging current: 500000
Max charging voltage: 5000000
Charge counter: 2417090
status: 2
health: 2
present: true
level: 83
scale: 100
voltage: 4184
temperature: 299
technology: Li-ion
fastChargeStatus: false
`
3. plugged, **not** charging (did that before forcing to charge through 'acc -e'):
`:/ $ su -c dumpsys battery
Current Battery Service state:
AC powered: false
USB powered: false
Wireless powered: false
Max charging current: 0
Max charging voltage: 0
Charge counter: 2415756
status: 3
health: 2
present: true
level: 83
scale: 100
voltage: 4140
temperature: 293
technology: Li-ion
fastChargeStatus: false
`

VR-25, 2019-07-08:
Since the first output is exactly the same as the last, this is just as unreliable as `acpi -a`.

This also means the latest prerelease won't work for you, @elem089.

It's one of those things that only work for some (it does for me).

Perhaps, instead of killing the potential usability of acpi, I will write a testing algorithm - so that `acpi -a` is only used if it actually works.

At least we're not back to ground zero (more like ground 0.5).

Thanks!

elem089, 2019-07-08:
Understand: you expected to see some kind of status change between "unplugged" and "plugged, not charging"
Tonight I can run that test  on my 2 well-aged Xperias, unless you feel that this is useless.
> This also means the latest prerelease won't work for you
Installed it anyway (before I started testing). But shouldn't do any harm, so I'll leave it at that

VR-25, 2019-07-08:
@elem089, reports on other devices will definitely be helpful - only in regards to `acc -a`, though. Forget about `dumpsys`.

Thanks

eugenesan, 2019-07-08:
Nexus 5X (acpi -a):
---------
* un plugged:
```
Adapater 0: off-line
```
* plugged charging:
```
Adapater 0: on-line
```
* plugged not charging:
```
Adapater 0: off-line
```

Pixel 3a (acpi -a):
---------
* un plugged:
```
Adapater 0: off-line
Adapater 1: off-line
Adapater 2: off-line
```
* plugged charging:
```
Adapater 0: off-line
Adapater 1: off-line
Adapater 2: on-line
```
* plugged not charging:
```
Adapater 0: off-line
Adapater 1: off-line
Adapater 2: off-line
```

elem089, 2019-07-09:
**Sony Xperia Tablet Z (pollux_windy), LineageOS 15.1:**
1. unplugged:
```
pollux_windy:/ $ su -c acpi -a
Adapter 0: off-line
Adapter 1: off-line
```
2. plugged (AC), charging enabled:
```
pollux_windy:/ $ su -c acpi -a
Adapter 0: off-line
Adapter 1: on-line
```
3. plugged (AC), charging disabled:
```
pollux_windy:/ $ su -c acpi -a
Adapter 0: off-line
Adapter 1: on-line
```

elem089, 2019-07-09:
**Sony Xperia Z1 compact (amami), LineageOS 16.0:**

1. unplugged:
```
:/ # acpi -a
Adapter 0: off-line
Adapter 1: off-line
Adapter 2: off-line
```
2. plugged (AC), charging enabled:
```
 :/ # acpi -a
Adapter 0: on-line
Adapter 1: off-line
Adapter 2: off-line
```
3. plugged (AC), charging disabled:
```
127|:/ # acpi -a
Adapter 0: on-line
Adapter 1: off-line
Adapter 2: off-line
```

elem089, 2019-07-10:
**Sony Xperia Z1 Compact (amami), LineageOS 16.0, acc build 20190709:**
- unplugged:
```
:/ # acpi -a
Adapter 0: off-line
Adapter 1: off-line
Adapter 2: off-line
```
- plugged, charging:
```
:/ # acc -e
(i) Charging enabled
:/ # acpi -a
Adapter 0: on-line
Adapter 1: off-line
Adapter 2: off-line
```
- plugged, not charging:
```
:/ # acc -d
(i) Charging disabled
:/ # acpi -a
Adapter 0: on-line
Adapter 1: off-line
Adapter 2: off-line
```

elem089, 2019-07-10:
**Sony Xperia TabletZ (pollux_windy), LineageOS 15.1, acc build 20190709:**
- unplugged:
```
pollux_windy:/ # acpi -a
Adapter 0: off-line
Adapter 1: off-line
```
- plugged, charging:
```
pollux_windy:/ # acc -e
(i) Charging enabled
pollux_windy:/ # acpi -a
Adapter 0: off-line
Adapter 1: on-line
```
- plugged, not charging:
```
pollux_windy:/ # acc -d
(i) Charging disabled
pollux_windy:/ # acpi -a
Adapter 0: off-line
Adapter 1: on-line
```

VR-25, 2019-07-10:
@eugenesan, unfortunately none of these two devices would support the plug in event listener based on `acpi -a`.

Poor driver implementations...
I'll keep looking for acpi alternatives.

Thanks

VR-25, 2019-07-10:
@elem089, that's great!

Just like Motorola, Sony got us covered on this.

Both Xperia devices support what I have in mind.

- Unplugged (external power supply status: _disconnected_ )

- Plugged in, charging (external power supply status: _connected_ )

- Plugged in, not charging (external power supply status: **still** _connected_  - here's where other devices fail)

Thanks

eugenesan, 2019-07-11:
> @eugenesan, unfortunately none of these two devices would support the plug in event listener based on `acpi -a`.
> 
> Poor driver implementations...
> I'll keep looking for alternatives.
> 
> Thanks

@VR-25 
I understand.
Thank you for your efforts.
Maybe one day we will find the solution.
Maybe one day

closed 2020-03-09 by VR-25