VR-25/acc #5: [Guide] Reporting Issues

VR-25, 2019-06-17:
1. Describe the issue in details. Include what changed/happened just before it started, and how you tried to solve the problem.
2. Add suggestions and/or extra (relevant) comments.
3. Attach the output file of `acc --log --export`.

elem089, 2019-07-13:
using acc on 3 different devices (currently latest dev acc-201907120); am using Terminal Emulator app by Jack Palevich (jackpal.androidterm):
most times (in rare cases it does not happen) I use one of the start commands (like acc -D start / restart or accd), then try to exit out of su the terminal app freezes. No matter how long I wait, there's nothing more happening in the terminal window. Only way to get out is to force-kill the app. It does not matter whether I stopped acc before the start or restart or not.
Same applies if I'm using the built-in terminal app on one of my two LineageOS devices (on my OP6 no such terminal app exists).
Added a screenshot and also a log both coming from one of my LineageOS driven Xperia devices; although I don't know whether the log is of any use because of course I had to kill the terminal then restart it to be able to create a log.
BTW: I had to zip the log as github does not allow to upload a .bz2 file...
[image]
[acc-logs-amami.tar.zip](https://github.com/VR-25/acc/files/3388806/acc-logs-amami.tar.zip)
Greets,
Lothar

closed 2021-07-22 by VR-25

VR-25, 2023-12-23:
[acc_v2023.11.26-dev_202311260_1001.zip](https://github.com/VR-25/acc/files/13758299/acc_v2023.11.26-dev_202311260_1001.zip)