VR-25/acc #7: Switch does not work if phone is not connected to usb

niander, 2019-06-28:
I was wondering if this is a particular problem with my phone or `acc` is expected to behavior this way.

If I run `acc -d` or `acc -e` with the phone disconnected from power source (usb) the new state is not saved even though I get no error messages. If the phone is connected everything works just fine. I have been using `acc` with Tasker so that is why I use mainly `acc -d` and `acc -e`.

I have a OnePlus 5t -Android 9, OxygenOS 9.0.4

VR-25, 2019-07-08:
It may be a particular problem with your device.
I'm investigating.

Meanwhile...
- Which acc version are you using?
- Try the latest prerelease: https://github.com/VR-25/acc/releases/tag/201907080

niander, 2019-07-10:
I am using the 2019.6.25

I will try the prerelease

niander, 2019-07-10:
I installed the prerelease but I am still getting the same behavior

Here some steps to reproduce the bug:
1. After enabling charge with `acc -e`, when I run `acc -i` I see that `INPUT_SUSPENDED=0`
2. I unplug the charging cable
3. The `INPUT_SUSPENDED=0` continues like that
4. I run `acc -d` and get a successful message: `(i) Charging disabled`
5. I run `acc -i` and I still get `INPUT_SUSPENDED=0`
6. I plug the charger again and it is charging.

If I follow the steps above but without removing the charging cable before `acc -d` the status of `INPUT_SUSPENDED=0` does change to `INPUT_SUSPENDED=1` and the charging is successfully suspended.

VR-25, 2019-07-13:
Hi,

While you can still get `mcs` and even `cs` (older) from this repository (by digging the commit history), I don't recommend it.

Here's what's actually happening and how to solve these issues:
- `chargingSwitch` is unset, meaning acc uses whatever switch works (i.e., it cycles through all - and **if one fails, its value is restored to default**).
- Upgrade to the latest pre-release: https://github.com/VR-25/acc/releases/tag/201907120
- Run `acc -s s` to enforce `input_suspend`.

closed 2019-07-21 by VR-25

niander, 2019-08-14:
Thanks for not saying it earlier but your solution worked. Thanks!