VR-25/acc #8: acc -D start / restart can freeze terminal window

elem089, 2019-07-13:
using acc on 3 different devices (currently latest dev acc-201907120); am using Terminal Emulator app by Jack Palevich (jackpal.androidterm):
most times (in rare cases it does not happen) I use one of the start commands (like acc -D start / restart or accd), then try to exit out of su the terminal app freezes. No matter how long I wait, there's nothing more happening in the terminal window. Only way to get out is to force-kill the app. It does not matter whether I stopped acc before the start or restart or not.
Same applies if I'm using the built-in terminal app on one of my two LineageOS devices (on my OP6 no such terminal app exists).
Added a screenshot and also a log both coming from one of my LineageOS driven Xperia devices; although I don't know whether the log is of any use because of course I had to kill the terminal then restart it to be able to create a log.
Let me add that this is the only command I know of that is able to freeze the emulator.
BTW: I had to zip the log as github does not allow to upload a .bz2 file...

VR-25, 2019-07-16:
The issue has to do with _double fork_ : e.g. `(sleep 120 &) &`. For some reason, this doesn't behave entirely as expected.
I'm looking for workarounds.

Thanks

elem089, 2019-07-19:
most probably you're aware: still the same with V 201907180

VR-25, 2019-07-20:
Yes, the issue hasn't gone yet.
Here's a workaround I figured out: `(su -c accd &)`

elem089, 2019-07-20:
I'm sorry, but I don't get it: what is `accd &` supposed to do? Do I need the brackets?
When I run it simply in su mode, accd & just returns a number like
`[1] 28446`
Then if I hit ENTER again response is
`[1] + Done   \accd`
Just entering `(su -c accd &)` or `su -c accd &` doesn't seem to do anything, at least I don't see any reaction.
And: I prefer to enter su mode alltogether, i.e. not using `su -c` with each command (being lazy, that's less typing)

VR-25, 2019-07-20:
It's really `(su -c accd &)`, not `su -c accd &`.
While the second also works, exiting the shell may require you to issue `exit` _twice_, or press <enter> once before issuing exit.

Note, you must be a regular user - meaning, this doesn't work is you're already root.

```
$ (su -c accd &)
$ exit
``````
$ su -c accd &
$ <enter>
$ exit
```
I'm a big fan of lazy typing.
I would add `s() { (su -c "$*" &); }` to Termux's `~/.bashrc`, so that I only need to type `s accd` to run `(su -c accd &)`.

elem089, 2019-07-22:
Alright, I'll try that next time

closed 2020-03-09 by VR-25