VR-25/acc #9: Pre-release 201907120: acc log not working

elem089, 2019-07-13:
just tried acc -l -a cat, as well acc --log --acc cat. Results in
`/sbin/acc[626]: -a not found`
or
`/sbin/acc[626]: -acc not found`
Device is OP6. Tested both with on-device terminal emulator and windows command through adb shell

VR-25, 2019-07-16:
Fixed

If there are no more major issues, the next release (includes this fix and more) will be _stable_.

Thanks

elem089, 2019-07-20:
ran my first test with this feature: no more error messages, but no log either:
`acc -l -a` opens a vim session in terminal with many empty lines and one last line saying
`/sbin/.acc/acc-OnePlus6.log 1/1 100%`
acc -l -a cat ends up in
`cat: can't open '/sbin/.acc/acc-OnePlus6.log': No such file or directory`
What am I doing wrong?

VR-25, 2019-07-20:
https://github.com/VR-25/acc/issues/10#issuecomment-513483961

closed 2019-07-21 by VR-25