VR-25/acc #10: Observations regarding older Sony Xperia devices

elem089, 2019-07-14:
Not really an issue, more a quick write-up regarding voltageLimit settings on my devices (Sony Xperia Tablet Z/pollux_windy 2014, running LineageOS 15.1; Sony Xperia Z1compact/amami 2014, running LineageOS 16.0; OP6/enchilada running OOS 9.0.7). Installed latest pre-release (201907120) on all three. Over the past few days tested vlotageLimit settings:
**pollux_windy** apparently has no file allowing voltage limitation, so I can rule this one out completely.
**amami** has a single file **usb/voltage_max**. I tried using that file and limiting it to 4000mV. Apparently this does not work: device is being charged with approx. 4200mV. This command:
`acc -v 4000`
is answered with a message that I'm probably using a wrong file.
**enchilada** seems to work: calling
acc -t --
tells me there are 5 files I could potentially use. After running some tests I concluded that I should use bms/constant_charge_voltage. I've currently set it to 3950mV, and it seems to work.
Let me know whether you're interested in some more infos regarding these devices.

VR-25, 2019-07-16:
Interesting

On any of these devices, is battery charge/capacity held for a long time while plugged in, not charging?

elem089, 2019-07-18:
Did over last night, i.e.: plugged in both devices, then issued 'acc -d', made sure that charging had stopped, then left them in that state for well over 5 hours. Interesting result: both devices behaved differently:
**pollux_windy:** 
when I started the battery indicator showed 49%. After 5 hours the battery was still at that value. Interestingly it had never entered deep sleep, instead had set at the lowest possible frequency. The system's battery page said "plugged in, can currently not be charged"
[image]

**amami:**
when I started battery indicator showed 74%, After 5 hours the battery had dropped to 67%. Also here CPU had never dropped to deep sleep but most of the time remained at the lowest possible rate. The system's battery page said "not charged"
[image]

2 more observations: 
- even after issuing the 'acc -d' command both devices show the "charging" indication in the statusbar; this is different with my OP6 where the charging indicator is switched off
- right after issuning 'acc -d' the daemon shuts down. I don't know if this is a wanted behaviour, but it's ok, I guess. What came as a surprise for me is the fact that after issuing 'acc -e' the daemon stays STOPPED. I tested this on all 3 devices, and it's true for all of them. Is this the expected behaviour?

VR-25, 2019-07-19:
Very Interesting

**pollux_windy**
- What's the _battery status_ reported by `acc -i` when plugged, not charging?

acc `-e` and `-d` stop the daemon to keep it from interfering (i.e., by turning charging back on/off).

These are essentially part of the _manual mode_, where the user decides exactly when to enable/disable charging. For the purpose of this tests, this is desirable.

Alternatively, you could use `acc pause resume` (e.g., `acc 80 70`). This keeps the daemon running.

elem089, 2019-07-19:
> What's the battery status reported by acc -i when plugged in, not charging?

```
pollux_windy:/ $ su -c acc -d

(i) Charging disabled

pollux_windy:/ $ su -c acc -i

NAME=battery
STATUS=Not charging
CHARGE_TYPE=N/A
HEALTH=Good
PRESENT=1
TECHNOLOGY=Li-ion
VOLTAGE_MAX_DESIGN=4200000
VOLTAGE_MIN_DESIGN=3200000
VOLTAGE_NOW=3933846
CAPACITY=66
CURRENT_MAX=6740673
CURRENT_NOW=1600
TEMP=230
CHARGE_FULL_DESIGN=6200000
CHARGE_FULL=6192000
CHARGE_NOW=-19075
```
Thanks for explaining acc -d / acc -e; makes sense then
Have a nice weekend

VR-25, 2019-07-20:
Strange
`STATUS=Not charging` means `pollux_windy` does in fact support `battery idle mode`.
Could you run the test in debug mode (`acc -x -t --`) and share `/sbin/.acc/acc-pollux_windy.log`?
Thanks

elem089, 2019-07-20:
weird:
```
d:\Android\acc>adb shell
pollux_windy:/ $ su
pollux_windy:/ # acc -x -t --

du: /sbin/.acc/acc-pollux_windy.log: No such file or directory
/sbin/acc[485]: [: 50: unexpected operator/operand

(i) [/sys/module/pm8921_charger/parameters/disabled 0 1] works
- battIdleMode=true
```
But there is a file acc-pollux_windy.log at /sbin/.acc/ with a modify time stamp of 2019-07-20 15:24, which must be the one just created. Attached it here
[acc-pollux_windy.log](https://github.com/VR-25/acc/files/3413655/acc-pollux_windy.log)

VR-25, 2019-07-20:
https://github.com/VR-25/acc/issues/9#issuecomment-513453788
That's the same problem - a minor bug.
A new release will be up once I finish working on a few other enhancements.
Thanks

P.S., `pollux_windy` does support `battery idle mode`. The output and log you provided confirm that.

elem089, 2019-07-21:
Just to be complete: ran the same test on my other xperia (amami): am I correct in interpreting the output that here battIdleMode is NOT supported?
```
d:\Downloads\Android\acc>adb shell
amami:/ $ su
amami:/ # acc -x -t --

du: /sbin/.acc/acc-amami.log: No such file or directory
/sbin/acc[485]: [: 50: unexpected operator/operand

(i) [battery/charging_enabled 1 0] works
- battIdleMode=false

(i) [/sys/devices/qpnp-charger-14/power_supply/battery/charging_enabled 1 0] works
- battIdleMode=false

amami:/ # ls -l /sbin/.acc/acc-amami.log
-rw-rw-rw- 1 root root 4254 2019-07-21 13:57 /sbin/.acc/acc-amami.log
```
[acc-amami.log](https://github.com/VR-25/acc/files/3414561/acc-amami.log)

VR-25, 2019-07-21:
Indeed, it doesn't work.
This is dictated by the kernel - yes, a different kernel may support that.

Tip: you can use a voltage control file as charging switch. It works almost like a switch that supports _battery idle mode_.
```
acc -s s battery/voltage_max DEFAULT_V TOO_LOW_TO_CHARGE_V
acc -s s battery/voltage_max 4380000 3500000
```

elem089, 2019-07-23:
Hmm - I assume that I must use a charging switch that already exists on my device, which in this case would be usb/voltage_max.
I tried `acc -s s usb/voltage_max 4350000 4000000`; how can I tell whether this is doing anything at all?
In addition, `acc -i` returns those two values among others:

> VOLTAGE_MAX_DESIGN=4350000
> VOLTAGE_MIN_DESIGN=4300000

Doesn't that mean that I cannot got below 4,3V no matter what I try?

closed 2020-03-09 by VR-25