VR-25/acc #11: acc log not working

cxyzzz, 2019-07-20:
acc -L used the tail is system tail not the busybox,it's dosen't support -F  only support -f

cxyzzz, 2019-07-20:
maby is my wrong,im using bash,PATH=/sbin:/su/bin:/su/xbin:/system/bin:/system/xbin
i see the acc.sh already rewrite PATH=/sbin/.magisk/busybox:$PATH or PATH=/sbin/.core/busybox:$PATH
i don't know why log not working.

elem089, 2019-07-20:
Guess I'm seeing a similar thing on one of my devices (Sony Xperia TabletZ / pollux_windy) when opening the log window from acca V 1.0.12; [see also here](https://forum.xda-developers.com/showpost.php?p=79928760&postcount=2991)

VR-25, 2019-07-20:
As you already noticed, @cxyzzz, the busybox tail is supposed to override the default - which doesn't support `-F`.
An explanation would be that your systems forbid PATH prepend.
Could you test that and post the output?
`su
echo $PATH
PATH=/dev:$PATH
echo $PATH
`

cxyzzz, 2019-07-20:
> As you already noticed, @cxyzzz, the busybox tail is supposed to override the default - which doesn't support `-F`.
> An explanation would be that your systems forbid PATH prepend.
> Could you test that and post the output?
> 
> ```
> $ su
> # echo $PATH
> # PATH=/dev:$PATH
> # echo $PATH
> ```

``` sh
$ su
# echo $PATH
/sbin:/su/bin:/su/xbin:/system/bin:/system/xbin
# PATH=/dev:$PATH
# echo $PATH
/dev:/sbin:/su/bin:/su/xbin:/system/bin:/system/xbin
```
It looks normal

cxyzzz, 2019-07-20:
i modify the acc.sh
``` sh
-   -L|--logwatch) tail -F ${modPath%/*}/accd-*.log;;
+  -L|--logwatch) echo $PATH && tail -F ${modPath%/*}/accd-*.log;;
```
then 
``` sh
# acc -L
/sbin:/su/bin:/su/xbin:/system/bin:/system/xbin
tail: not integer: -F
```
It seems that the path has not been modified

VR-25, 2019-07-20:
Oh, I see.
Do you have busybox installed?

cxyzzz, 2019-07-20:
@VR-25 ye,I install it at /system/xbin

cxyzzz, 2019-07-20:
log working normally after uninstalling busybox.thanks !

closed 2019-07-20 by cxyzzz

VR-25, 2019-07-20:
As I presumed.
I'm improving the busybox handling mechanism to prevent this and similar issues.
Thanks