VR-25/acc #13: ACC doesn't stop charging on plug

Evengard, 2019-07-28:
This is a weird case.
Let's assume the battery level is 90%. The config is:
```
capacity=0,101,80-85
temperature=100-105_90
```
all other config settings are default.

When I connect the phone to a charger (at battery level 90%), Android says that the battery isn't charging...
But in reality, when I check again with another application (in my case AccuBattery Pro), it is *actually charging*! Android doesn't show it up as charging but it is still charging. Even more, after a while the percentage increases to 91, 92 and eventually 100%.

When I do acc -t it stops charging as expected. When I reconnect the cable after that, it charges till 100% again.

Tried every single one switch present in `acc -s s:`. Same with all of them.

Asus Max Pro M1 (X00T), Lineage OS 16.0 (official nightlies, but I remember it was the same on stock ROM too).

I've zipped the tar.bz2 file to allow it's upload here on github.
[acc-logs-ASUS_X00T_6.tar.zip](https://github.com/VR-25/acc/files/3438854/acc-logs-ASUS_X00T_6.tar.zip)

VR-25, 2019-08-13:
If I read it right, could you confirm that `acc -d` _does_ in fact turn off charging as expected - but the change is overridden on replug?

Evengard, 2019-08-13:
Yes, I can confirm that. Just tried out - on replug it just keeps charging. And even more Android still thinks it doesn't charge, but the charging is ACTUALLY happening.

closed 2020-03-09 by VR-25

Evengard, 2020-03-09:
Why did you closed it? The bug is still present, and a workaround would be very much appreciated.