VR-25/acc #17: Switches only work before the first usb cable disconnect

tkersten09, 2019-10-11:
I have an LG G5 H850 and installed acc via Magisk. I use the [UNOFFICIAL LineageOS-15.1 for G5](https://forum.xda-developers.com/lg-g5/development/rom-unofficial-lineageos-15-1-g5-t3806842) and the system is quite fresh.

But i realised that acc is only able to switch off the battery charging before the usb cable is disconnected for the first time after boot up. I include a gif which illustrates the process. 

The problem is repeatable like so: 

1. I reboot the phone without a usb cable connected. I wait for around 2 minutes to give all programs and the acc daemon to boot up. 
2. Then i start a shell with Termux. And use:
 `acc -t --`
to test which switch works. This fails because no cable is connected. So i connect the usb cable and then it reports that 2 out of 4 switches work. I can repeat the test several times showing the same result. 
3. I disconnect and reconnect the usb cable, then the test fails and reports that now all switches do not work. After that all switches do not work anymore until a reboot is done.

I include several logs i saved with:
`acc -l -e` [acc_logs.zip](https://github.com/VR-25/acc/files/3717218/acc_logs.zip)
[image]

VR-25, 2019-10-12:
Toggling multiple switches may trigger the issue.
Did you try using a single switch under such conditions? -- `acc -s s`

tkersten09, 2019-10-12:
I chose `battery/charging_enabled 1 0` as my switch. And the switch test `acc -t` now reports that my device is supported. And acc now stops charging at my cutoff battery charge, which is 90% for me. So technically my issue is resolved.

But there are still some weird things:
- If i do the switch test with a battery charge over my cutoff battery charge, like 92%, the test fails again. And this sticks until a reboot. But charging still ends, if the charge is 90% or higher. So everything works, but the switch test says otherwise. 
- What bugs me the most, is that if the charge is greater than 90%, charging still stops and the charge drops over time below 90%, but the little thunderbolt in the battery icon does not go away and the indicator led on my phone does not turn off (as it does if the charge hits 90% from below and cuts off charging and stays there). So something happens, but something else still goes wrong.

So everything is fine, if the battery charge approaches the cutoff charge from below and does not go over 90%. But it glitches out between 91% and 100%. Maybe this is already known. But if you are interested in this glitch, here my new logs:
[acc-logs-h1.tar.bz2.zip](https://github.com/VR-25/acc/files/3720989/acc-logs-h1.tar.bz2.zip)

And if you wish, i could commit some additions to the readme which recommends a reboot, if the switch tests fails. And suggests that choosing a switch might resolve the problem.

VR-25, 2019-10-13:
1. `battery/charging_enabled` is working.
2. `acc -t` is working with that control file.
3. `acc -t --` misbehaves, because it always tests **all** control files. Some of which were found to be problematic.
4. Charging indicators won't go off after disabling charging.
5. A reboot is required to re-enable charging (`rebootOnPause=timeoutSeconds`).

6. The log shows exit code 143, which suggests something is killing accd after it tries disabling charging a couple of times.

7. `Assuming you have a specific charging switch set, is `acc -t` prone to the inconsistent behavior of `acc -t --`? Recall that the first toggles only the set switch, while the second always toggles all.

8. If you don't perform any test, and the control file is still `battery/charging_enabled`, does accd work as expected?

P.S., there's a [new dev build](https://github.com/VR-25/acc/releases/tag/201910130/) available. It's ready for daily use. If interested, _after_  reading the changelog, you can upgrade with `acc -u dev`.

closed 2020-03-09 by VR-25