VR-25/acc #20: Not working on the Google Pixel 4

airbourn, 2019-11-21:
ACC doesn't seem to be working on the Pixel 4 (eg. doesn't stop charging when it reaches the pause charging level). Also tried the `forceStatusAt100=5` workaround - still nothing. Currently on the stock kernel. 

[acc-logs-flame.tar.zip](https://github.com/VR-25/acc/files/3874478/acc-logs-flame.tar.zip)

jpcaldwell30, 2019-12-04:
I have the same issue as the user above. I have a pixel 4 XL running the latest version of android 10 and after some testing it seems that none of the switches work which is why we are unable to get the service to do anything. After some digging i think its because the phone has different control files/missing ones so we just need to find one that works as a switch. Any ideas on where to start?

Aiakio, 2019-12-06:
Also same issue here. Pixel 4 XL.

VR-25, 2019-12-14:
I found the following new charging switches:
```
/sys/devices/platform/soc/c440000.qcom,spmi/spmi-0/spmi0-02/c440000.qcom,spmi:qcom,pm8150b@2:google,bms/power_supply/charger/charge_disable 0 1

/sys/kernel/debug/google_charger/chg_suspend 0 1

/sys/kernel/debug/google_charger/input_suspend 0 1
```
Pro tip: the first can be abbreviated as `charger/charge_disable 0 1` because it matches `.*/power_supply/.*`.

Test command: `acc -t [file] [on] [off]`
  e.g. `acc -t charger/charge_disable 0 1`

Set a working switch with `acc -s chargingSwitch [file [on] [off]`
  `acc -s chargingSwitch charger/charge_disable 0 1`

jpcaldwell30, 2019-12-14:
Thank you! The first one works for me on the Pixel 4 XL. I haven't tried any of the others but I suspect they will as well. Can we expect this to be included in the next app update? Also, could you add the ability to add a custom switch in the app in the select switch popup?

VR-25, 2019-12-14:
Yes, the next version will include these changes.

airbourn, 2019-12-15:
@VR-25 Now working on my Pixel 4 as well after trying the commands above. Thanks a lot!

closed 2020-02-25 by VR-25