VR-25/acc #23: The on and off values of the charging switch cannot contain spaces

kbios, 2020-01-08:
I have a device where the charging switch requires two numbers separated by a space. Acc doesn't currently support this because it uses awk '{print $whatever}' to extract the on/off values in various places, so quotes/escape sequences don't work. I managed to get it working by using read instead, like this:
```
local switch=""
read -A switch <<< $(get_value chargingSwitch)
file=${switch[0]}
value=${switch[2]}
```
This way I can escape the space in the config file (eg `chargingSwitch=/foo/bar 0\ 0 1\ 1`) and everything works.

VR-25, 2020-01-09:
A while ago, I came across `/proc/mtk_battery_cmd/current_cmd "0 0" "0 1"`. Is that your switch as well?

ACC is undergoing major changes. As part of these, several config variables are now arrays (e.g., `chargingSwitch=(/proc/mtk_battery_cmd/current_cmd "0 0" "0 1")`.

If you want to try experimental builds, message me on t.me/vr25xda, fb.me/vr25xda or m.me/vr25xda .

kbios, 2020-01-09:
Yep, that's the switch (btw it's a Blackview 9600 Pro).
One unrelated thing I've noticed is that toggling it stops charging but keeps the battery at that level, while to make it actually discharge I need to toggle `en_power_path` as well.
As for trying the experimental builds, thanks a lot for the offer, but unfortunately I cannot tinker much with this device as it is the primary phone of a friend :(

VR-25, 2020-01-09:
I can't get enough of Mediatek's "nice" surprises. Holly nuts!

Could you elaborate more on `en_power_path`? I can make acc toggle it.

kbios, 2020-01-09:
So `/proc/mtk_battery_cmd/en_power_path` is normally 1, when set to 0 it stops charging and also begins discharging but there is a catch: the battery status always remains "charging" , so acc doesn't work.
To get the complete behaviour (battery stops charging, starts discharging and reports the correct status) one has to toggle both `en_power_path` and `current_cmd`.

VR-25, 2020-01-09:
Interesting and Intriguing!
This will give birth to a new feature: toggling multiple switches at once.

kbios, 2020-01-09:
Wow that's so cool! Manufacturers sure know how to make things interesting 😂

kbios, 2020-01-12:
Just to make things more fun I found out that if `current_cmd` is toggled on while the device is not plugged in, the battery will report "charging". Yes really -_-

VR-25, 2020-01-12:
Terrible Design!

Lets say the device is plugged in and charging. After charging is disabled (i.e., through toggling the 2 switches), is it enabled automatically on replug, without toggling the switches back?

Were you able to find any file that reports the correct external power status (i.e., plugged/unplugged)?
Still on this question, does `su -c acpi -a` do that?

kbios, 2020-01-17:
Sorry for the delay, I managed to get some time with the phone to run the tests. The results are braindead haha
- Once one touches current_cmd it becomes fully manual, ie the phone won't charge when plugged in unless the switch is toggled
- acpi -a reports two adapters, "Adapter 0" seems to reliably indicate whether the phone is plugged in while "Adapter 1" is always "Off-line"

VR-25, 2020-01-17:
Finally, at least one pleasant surprise... hopefully! :grin:

If `acpi -a` really reports the correct external power supply status (plugged/unplugged), regardless of whether switches were toggled, we have a winner.

From `acpi -a` and `battery/status`'s data combination, we have three basic statuses:
- Plugged in and charging (at least one adapter online AND battery/status == "charging")
- Plugged in, but not charging (at least one adapter online AND battery/status == "discharging|not charging")
- Unplugged

I just finished implementing and testing the "multiple switches toggling" feature. It's ready for third-party testing. I'm in touch with another MTK user - who will be testing the real deal.

kbios, 2020-01-17:
That's great! So to ensure that the braindead status "unplugged but charging" never happens I guess acc will have to toggle the switches when it detects an unplug haha

VR-25, 2020-01-17:
_Thou Shalt Go Mad._ 
- MediaTek Inc.

I started pouring extra gasoline onto rogue neurons to come up with suitable algorithms.

The basic idea is implementing stricter "toggling rules" - so that the "unplugged but charging" madness never happens in the first place. This is possible when `acpi -a` reports real adapter status.

VR-25, 2020-01-20:
I implemented everything we discussed so far.
Even though this is a pre-release, the daemon is pretty much stable at this point.

https://www.dropbox.com/s/cdfok5fek2g5z5f/acc-202001180.zip?dl=0

Notes

- Don't use `acc -s|--set` to edit config. It's not ready yet. Use _acc -c|--config EDITOR OPTS_ instead (e.g., _acc -c vim_). If you run just `acc -c`, the fallback editors are _nano -l > vim > vi_. You can always edit _/data/adb/acc-data/config.txt_ with any other means.

- Apparently, MTK requires a [slightly] longer delay after toggling the switches - for changes to take effect. ACC has the config variable _switchDelay_ (default: 1 second). You may want to increase that before doing any testing - and play with it along the way. This is because in auto mode (chargingSwitch unset), the switch cycle is in charge. After switchDelay, it resets switches that don't work. That's where the default delay can be an issue. switchDelay=10 is a good starting point.

- If you don't want to play with switchDelay right now, set `chargingSwitch=(/proc/mtk_battery_cmd/current_cmd "0 0" "0 1" /proc/mtk_battery_cmd/en_power_path 1 0)` and test whether `acc -d` disables charging - and `acc -e` re-enables it. This is the manual/daemon-less mode.

- The documentation is not fully updated yet. At this point, besides switchDelay and chargingSwitch[], the only other values you have to care about are capacity[2] (resume, default: 75) and capacity[3] (stop, default: 80).

kbios, 2020-01-22:
Wow that's great! I can't wait to try it, as soon as I see my friend I'll try and report back

kbios, 2020-02-03:
So I finally got the chance to test (sorry for the delay!), but I'm running into a weird problem: I flashed the zip with magisk but apparently it gets installed into /sbin/.magisk/modules/acc/ but not into /sbin/.acc/, and this prevents it from starting... if i try manually I get like
```
Blackview:/ # /sbin/.magisk/modules/acc/acc.sh                                                                                                                   

/sbin/.magisk/modules/acc/acc.sh[466]: .: /sbin/.acc/acc/setup-busybox.sh: No such file or directory
```
I'm probably missing something obvious sorry

VR-25, 2020-02-25:
@kbios, the repo is up-to-date.
Dowload the latest version: https://GitHub.com/VR-25/ACC/archive/dev.zip .
Extract the zip and run the following to install:
$ su
# sh /path/to/extracted/install-current.sh

closed 2020-02-25 by VR-25