VR-25/acc #25: Charging stuck at 70%

unl0cks, 2020-02-17:
After I installed this my phone only charges to 70% and I haven't found a way to revert this. My battery is already bad and I need the extra 30%. I used some acc commands and accidentally made it worse, so it only charges to ~61% now. I uninstalled the module at one point but it didn't revert the changes this module had applied. What commands do I need to use to fix this? This is my current config:

versionCode=201910130
capacity=0,100,100-100
capacityOffset=+0
capacitySync=false
temperature=70-80_90
coolDownRatio=
resetBsOnPause=false
resetBsOnUnplug=false
loopDelay=10,15
chargingSwitch=
applyOnBoot=battery/charge_control_limit 0
applyOnPlug=battery/charge_control_limit 0
maxChargingVoltage=
rebootOnPause=
chargingOnOffDelay=1
language=en
wakeUnlock=
prioritizeBattIdleMode=false
forceStatusAt100=

VR-25, 2020-02-18:
`capacity=0,100,100-100`
`battery/charge_control_limit 0`

These lines...
What are you using acc for?

All changes are supposed to be volatile, like the RAM itself.
Changes are made in /sys, which rests in RAM.
All RAM contents are gone as soon as the power is cut off (e.g., on shutdown/reboot)

Every once in a while, someone reports such a weird issue - which apparently happens on devices with unusual driver setups. Perhaps these drivers have some kind of persistent storage.

Anyway, the exact reason behind this weirdness is still unknown to me.

You can try a few things:

1) `acc -e` (enable charging)
2) Forced reboot
3) Shutdown
4) Boot in safe mode
5) Disconnect and reconnect the battery from the motherboard, if feasible
6) Clear cache
7) Reflash firmware

unl0cks, 2020-02-23:
> `capacity=0,100,100-100`
> `battery/charge_control_limit 0`
>
> These lines...
> What are you using acc for?
>
> All changes are supposed to be volatile, like the RAM itself.
> Changes are made in /sys, which [...]

I'm using acc to extend my battery life. My battery drains very fast since it's old, but since the battery charge is limited to 70% it's hard to fix. I have already tried the following:

1. Using the `acc -e` command (aswell as `acc -e 100%`).
2. Clearing partition cache and dalvik.
3. Rebooting.
4. Uninstalling acc.
5. Changing capacity.
6. Uninstalling magisk (this worked but I need root for various reasons (boosting performance, battery life, etc).
7. Reflashing custom ROM (PixysOS 3.03).
8. Using the `acc -f` (aswell as `acc -f 100%`)

Nothing seems to work. I have a Motorola Moto G(5S) Plus (sanders) running PixysOS 3.03 (Android 10). Isn't there any command to limit the battery to something else (atleast 90%, but preferably 99-100%)?

Thanks in advance.

VR-25, 2020-02-24:
Hmm, if I read that right, it's strange that uninstalling Magisk works... 🤔

Anyway, do `acc -i` and `dumpsys battery` report the same battery level?

unl0cks, 2020-02-24:
@VR-25 I fixed the problem by removing all "optimization" modules and EdXposed. I don't know which of them fixed it.

closed 2020-02-25 by VR-25