VR-25/acc #28: [le zl1] Sometimes charging icon stuck on & battery saver stuck disabled

jesse-git, 2020-03-11:
I uploaded power supply log for device le_zl1 on the dropbox.  generally ACC works okay, but roughly around 1 of every 5 times when charging, when the charge stops at the designated percent, a weird thing happens:
[LIST]
[*]  lightning bolt charge icon remains
[*] battery saver mode will not reengage, automatically or manually
[/LIST]

it does  fix itself  upon reboot.  It happens on:

[LIST]
[*] RR-P-v7.0.2-20190710-zl1-Official  with black screen kernel 6.6.1
[*] aicp_zl1_p-14.0-WEEKLY-20191226
[/LIST]

VR-25, 2020-03-15:
Hi,

1. Install the [latest](https://github.com/VR-25/acc/releases/) pre-release. It's stable enough for daily use.

2. Run `acc -t --` to see which charging switches work.

3. Enforce a working switch with `acc -s s`.

4. Test its reliability and repeat steps 3 and 4 as needed. For details on why this procedure is recommended, refer to https://github.com/VR-25/acc/blob/dev/README.md/#charging-switch .

jesse-git, 2020-03-18:
read through it, unfortunately did not help

```
# acc --version

2020.3.14-dev (202003140)
# acc -t --
(!) [dc/charging_enabled 1 0] won't work                      
(!) [parallel/charging_enabled 1 0] won't work                
(i) [battery/charging_enabled 1 0] works                      
- battIdleMode=false                                          
(!) [battery/battery_charging_enabled 1 0] won't work
(!) [battery/input_current_settled 0 1] won't work
```
switch_delay was default 1.5
`acc -s switch_delay=5` didn't help.

VR-25, 2020-03-30:
It turns out this issue is triggered by the kernel.
There's a new acc release.
1. Install it.
2. Run `acc -s sd=1.5; acc -t --%` to see which switches work.
3. Enforce the first with `acc -s s`.
4. Test its reliability. If it doesn't work as desired, try the next. Repeat as needed.

closed 2020-03-30 by VR-25

jesse-git, 2020-04-02:
@VR-25  unfortunately only the same charge switch is reported as before.  the behavior is the same.

 could you elaborate about the kernel or give a link? should I file a bug with them?