VR-25/acc #32: 202003301 accd exits soon after starting

jesse-git, 2020-03-31:
after upgrading to 2020.3.30-r1-dev (202003301)  from an older version a few days ago, `acc` then `start`  reports `started`, but in reality it stopped  due to unknown reason. Running `acc`  again shows that it is not running.

jesse-git, 2020-04-01:
I got accd  to start by specifying  the values manually in the configuration file, for `capacity` and one or more of `temperature`, `resetBattStats`, `loopDelay`, `chargingSwitch`. there seems to be an issue with assigning defaults if the variables in the configuration file are blank ( as in the default config file).

VR-25, 2020-04-01:
You're facing a very strange issue.
This only happens when the config is messed up.
The provided log archive doesn't even have config.txt.
Do you have any idea why the said variables are null?
As a general rule of thumb, no variable can be null. Thee are very few exceptions.
Did you try resetting the config (`/data/adb/acc-data/config.txt`) by removing it and restarting accd?

jesse-git, 2020-04-02:
I had briefly installed 20200330**0**,  maybe that is the reason?
At some point upgrading from stable to dev to more dev versions,  the config got wiped.  I forget if I deleted the config file  to start fresh or if acc did somehow.
 I never directly modified the config file until trying to fix this issue.  but now it seems to be fixed.

 unfortunately I did not save what I saw with the blank values in the configuration file. But the capacity line looked like this  before. in general, all variables with parentheses had the correct number of spaces but no values. And all variables without parentheses were just empty after the equal sign.
```
capacity=(     )
```

 deleting the configuration file now so that it writes the new default  file results in  half of the values being filled with the defaults and the other half being "blank".

```
configVerCode=202003301
capacity=(-1 60 70 75 +0 false)
temperature=(70 80 90)
cooldownCurrent=()
cooldownRatio=()
resetBattStats=(false false)
loopDelay=(10 15)
chargingSwitch=()
applyOnBoot=()
applyOnPlug=()
maxChargingCurrent=()
maxChargingVoltage=()
rebootOnPause=
switchDelay=1.5
language=en
wakeUnlock=()
prioritizeBattIdleMode=false
forceChargingStatusFullAt100=
runCmdOnPause=()
dynPowerSaving=0
```

 comparing that to my previous configuration,  I was still missing values for `switchDelay`, `language`, `dynPowerSaving`. 

 I assume the above configuration looks okay now?

VR-25, 2020-04-02:
I see.
It was an edge case.
I faced a very similar issue before.
The  new config is fine.

Just improved the config patching algorithm.
Thanks!

closed 2020-04-02 by VR-25