VR-25/acc #36: Asus P024 Lineage 14.1 not stopping charging

kinger79, 2020-04-05:
Hi, 

Charging not stopping on set margins 45 90. I have a few Asus P024 Lineage 14.1 working nicely and ACC would be greatly appreciated.

Br, Kinger

[acc-logs-P024.tar.gz](https://github.com/VR-25/acc/files/4433255/acc-logs-P024.tar.gz)

VR-25, 2020-04-08:
Hi,
Your kernel is to blame.
Install the latest version (recently uploaded) and test whether a cutoff value of 75 works.

Next, create a file somewhere and save the following in it:

`/sys/devices/soc.0/78b7000.i2c/i2c-3/3-006c/charge_status 0 1
/sys/devices/soc.0/78b8000.i2c/i2c-4/4-006a/charge_status 0 1
battery/temp_cool 100 999
battery/temp_warm 450 100`

Once done, plug the device, run `acc -t /path/to/file` and share the output.

kinger79, 2020-04-11:
Hi,
I've tried with **v2020.4.10-beta (202004100)**

Sharing the outcome: 

(i) This may take a few minutes...
(i) accd stopped

(!) [/sys/devices/soc.0/78b7000.i2c/i2c-3/3-006c/charge_status 0 1 
] won't work

(!) [/sys/devices/soc.0/78b8000.i2c/i2c-4/4-006a/charge_status 0 1 
] won't work

(!) [battery/temp_cool 100 999 
] won't work

(i) accd started

VR-25, 2021-07-21:
Test the upcoming release and report back.

closed 2021-07-22 by VR-25