VR-25/acc #39: Charging is resumed before battery level reaches the specified value

Tomoms, 2020-04-16:
Dear developer,
I'd like to report an issue that occurs both on the stable release from July 2019 and on the latest beta (April 10th).
It's very simple: if I, for example, run
`acc 55 60`
or apply a "55->60" profile with AccA, the battery is charged correctly until the level reaches the upper bound (in this case, 60), but charging is **not** inhibited until the level falls to 55. Instead, the battery starts charging again as soon as it drops to 59, or sometimes even before that: sometimes it resumes charging intermittently, for very short time intervals (less than a minute), when it's still at 60. Sometimes, this happens so often that these small charges manage to bump the battery level up beyond the upper bound, e.g. to 61.
I've tried completely removing acc and AccA and all the config files, and performing a fresh installation, but the issue is still there.
I'm using an Oneplus One, with LineageOS 16.
I've attached the logs (exported about one minute after the first "short charging" event).
I'm of course available for further discussion of the problem.
Thanks!

[acc-logs-A0001.tar.gz](https://github.com/VR-25/acc/files/4485321/acc-logs-A0001.tar.gz)

VR-25, 2020-04-16:
`acc [pause %] [resume %]` is the correct syntax.
You ran `acc 55 60` instead of `acc 60 55`.

Tomoms, 2020-04-17:
Gee, I feel so stupid! Anyway the issue persists. Also, when I was using AccA the pause/resume thresholds were set in the correct way and the issue still occurred.
I've tested the charging switches, only one out of two is working, and of course, I'm using that one. The old stable version used to recognize two working switches, but in the latest beta only one is detected, together with another (third) switch that however, doesn't work properly.

Some additional info: these unwanted short charges usually start when I wake the device from sleep. I'm not even sure they can be accounted as real charges, because the charging led of the device turns on, but the battery icon doesn't show the lightning it usually shows when the battery is charging. So maybe the kernel's charging driver is going mad, and I mustn't blame acc for this.

VR-25, 2020-04-17:
See if today's version makes any difference.

Tomoms, 2020-04-18:
Still happening. Do you need more logs?

bollocks-fks, 2020-04-29:
I think I'm having the same issue. I'm using a Redmi Note 6 Pro with LineageOS 17.1. Charging seems to be restarted regardless of the values of resume capacity and pause capacity. It seems to just be a matter of time. Then it seems like accd notices that this is wrong and turns it off. Over time this will fully charge my phone.

The bug happens for me with version 4.22-beta and 4.27-beta. 4.17-beta and 4.20-beta don't have this bug for me.

bollocks-fks, 2020-04-29:
I left 4.20-beta running over night with the charger plugged in and the bug came back. I'll leave 4.17-beta running for a while and see what happens.

bollocks-fks, 2020-05-01:
4.17-beta doesn't seem to have the bug after running it for about 2 days.

VR-25, 2020-05-03:
Hi,
I'm watching this closely.
A new version will be up soon.
Keep trying.
We're getting there.
While on this trial and error cycle, keep the following in mind: https://github.com/VR-25/acc/blob/dev/README.md/#charging-switch .

bollocks-fks, 2020-05-05:
4.29-rc works great for me. Tomoms seems to have noticed the issue on a different version, so maybe the bug is different on his hardware.

I've been just been using the default switches. acc -t reports more than one switch works. I don't understand the differences between them, so I just figured leave it be. Looking at the issues brought up in the readme:

Leaving the screen on didn't help when I tried it, but I didn't thoroughly test this across versions. It was just something I checked to see if the phone was going to sleep, and accd wasn't given the chance to run properly.

I don't know how to detect a wakelock being held. If there's a command I can run to test this, I'll run a test on switches if you want. I haven't noticed a battery drain problem. 

CPU usage of accd.sh as reported by ps seems to be 0.1% over the long term. I haven't noticed any change I figure that's pretty good. 

I have noticed Android rarely not reporting battery status correctly. The files in /sys/class/power_supply seem to be working fine when this happens, and a reboot clears it up. So I figure this is a bug in the ROM I'm using. This worked fine in LineageOS 16.1. I've also lost functionality in the charge counter file since the upgrade. They're still there, they just report 0. The capacity files are working fine when I run a script to watch and record them.

I'll give 5.4-rc a try now, and make sure it's still running smoothly.

Tomoms, 2020-07-08:
Hello, sorry for having been absent for a very long time.
I'd just want to say that acc v2020.6.16 is working perfectly and this bug is definitely fixed. I've just updated to the latest version (v2020.7.3), I'm pretty sure it'll work well too.
Thank you!
Greetings.

closed 2020-07-08 by Tomoms