VR-25/acc #47: Phone doesn't charge [Doogee N100 -MT6763]

Cyberavater, 2020-06-16:
Installed from f-droid. When I plug AC adepter it shows charging but the charge never goes up. I tried waiting for 6hrs.

VR-25, 2020-06-17:
MTK is problematic by design.
Flash the latest acc zip and see if the problem persists.

Cyberavater, 2020-06-18:
It seems Magisk has the latest version (update available on Magisk) while the one from F-droid is older. So I updated using Magisk. And it seems to be working fine now. (Why not keep the f-droid one updated too?) Also, I'd like to know if it slows down the fast-charging speed. (Not that I tested and found it disappointing or anything, I'd just like to know it from the dev himself).

Thanks for your time & wonderful app.

closed 2020-06-18 by Cyberavater

VR-25, 2020-06-18:
I forgot to mention that you have to update the [app](https://github.com/mattecarra/acca/releases/latest) as well.
I'm not the main developer of AccA.
The F-Droid version is lagging behind because the main dev has been very busy in the last few moths.
The COVID-19 situation has been slowing everything down.

Cyberavater, 2020-06-18:
Okay, so what about my other question?
> Also, I'd like to know if it slows down the fast-charging speed. (Not that I tested and found it disappointing or anything, I'd just like to know it from the dev himself).

Also, why not add the app in the Magisk module? #49