VR-25/acc #48: Large discrepancy between power_supply/battery/uevent and dumpsys battery

f8lxninja, 2020-06-17:
Hello everyone,

I have a Pixel 3 currently, and when I run acc -i I see a large disparity between the level reported from /sys/class/power_supply/battery/uevent and dumpsys battery. My question is, is this normal behavior? At first it was only 5% off, but now it is upwards of 15% off and appears to be getting worse. Attached is a screenshot of what is reported from acc -i. Note that this only started when the newer versions of ACC were released, starting with v2020.6.14.

[image]

I did look in the README for any information regarding Pixel devices and this issue but it seems to be missing. Any help would be appreciated!

VR-25, 2020-06-17:
Hi,
This is a classic Pixel issue.
Partial charges and discharges trigger it (not acc).
The issue was even reported to Google, several times and by different people.
They simply ignored all reports.
To work around that, acc keeps Android in sync with the real battery level (reported by the kernel).
When you run acc -i, the real values are shown.
Stop acc daemon and you'll see a drastic change on your battery indicator.
As long as acc daemon is running, you're covered.
The people at Google don't want to fix their stuff.
Live with it.

f8lxninja, 2020-06-17:
Ah okay, the weird part is that it would not get exponentially worse with time before the newer revisions. It stuck around 3% or so off and I could do an offset to make me happy. If ACC is correct all the time, then I'll stop worrying about it :). Thank you for the insight!

closed 2020-06-17 by f8lxninja