VR-25/acc #50: Charging switch automatically switches against user settings

PenguinDOOM, 2020-06-26:
Device: Pixel4 (Flame)
OS: Android 10
ACC Version: v2020.6.16

I don't know why, but after a while after switching the charge switch, it automatically switches to another switch.

The log was output after switching from `charger/charge_disable 0 1` to `/sys/kernel/debug/google_charger/chg_suspend 0 1`
[acc-logs-flame.tar.gz](https://github.com/VR-25/acc/files/4835121/acc-logs-flame.tar.gz)

VR-25, 2020-07-01:
ACC changes the charging switch when it fails to disable charging.
That settings was never meant to be static.
There are a couple of reasons for this behaviour.
The main is that different power supplies may use distinct charging switches.
A notable example is wireless charging.
It often requires its own switch.
Another reason is that some switches are not always reliable (i.e., fail to disable charging under certain conditions - e.g., screen off, capacity above a certain threshold and so on).
Thus, ACC responds accordingly.

blackwind, 2020-07-03:
I'm having the same issue. `battery/charging_enabled` works just fine for me, but `acc -t` and even `acc -t 30` disagrees. When I hit my maximum charge level, accd starts checking for a working switch (even after I've explicitly set it), doesn't find one, gives up, and exits.

I propose an option to disable this auto-switching functionality since acc apparently has trouble correctly detecting working switches on some devices, and the current behavior of giving up and exiting without notifying the user in any way should be improved somehow. I tried this module several months back, saw it did nothing, and immediately uninstalled. Same thing with ACCA -- installed, saw it did nothing, saw "daemon stopped", restarted it, saw "daemon stopped" again after hitting my charge limit again, and immediately uninstalled. Great intentions with the auto-switching functionality, but very, very poor user experience when things don't go as planned.

PenguinDOOM, 2020-07-18:
I have it set to stop charging at 85% but the charge switch switches to automatic and the acc daemon stops. This is tricky because on my pixel4, only `charger/charge_disable 0 1` works as expected.
I can't upload the log from smartphone, I'll upload it later.

PenguinDOOM, 2020-07-19:
Log
[acc-logs-flame.tar.gz](https://github.com/VR-25/acc/files/4943119/acc-logs-flame.tar.gz)

VR-25, 2020-07-19:
Install today's version and share the output of `acc -t`, along with a fresh log archive (`acc -le`).

PenguinDOOM, 2020-07-19:
The latest log.
[acc-logs-flame.tar.gz](https://github.com/VR-25/acc/files/4943986/acc-logs-flame.tar.gz)

VR-25, 2020-07-23:
The latest version has a new eatyre

PenguinDOOM, 2020-07-25:
I added `--` and ran it a few times and it works fine. Thanks.

closed 2020-09-24 by VR-25