VR-25/acc #51: error trying to make a call

pega2k, 2020-07-02:
Hello,

After installing acc v2020.6.14 (202006140), I can't not make any call, I installed it with Magisk 20.4 in a Redmi Note 7 MUI 12, Android 10 (xiaomi.eu rom 20.6.18). 

The error is "that I don't have any account registered to do that type of call".

If I disabled the module and reboot the phone, I can call again.

This is my first time using acc in that phone.

In the end I made it to work after doing the following procedure, without install or disable the acc from Magisk, I installed acca from f-droid, change the acc to the bundle one in acca (can't call), reboot the phone (can't call), and again switch to the last version v2020.6.14 from the acca and reboot, now I can call without problem and the acc work OK.

Please forgive me my bad english.
Greetings, Pedro Enrique.

ovizii, 2020-07-03:
I have the same problem coming from xiaomi.eu firmware stable version 11 I flashed to the stable 12.0.1 noticed I couldn't make calls and then tried 12.0.2 but still having this problem. AFter disabling the ACC Magisk module the problem is gone. 

I had ACC magisk module v2020.6.16 installed before and didn't notice this problem. Not sure if this exact version caused it as I very rarely use my phone for actual GSM conversations :-)

TinyRaindrop, 2020-07-04:
AOSP rom, after updating to v2020.6.16 faced the same error:
>call cannot be placed because there are no calling accounts

Rolled back to v2020.5.4-rc (bundled in AccA), rebooted and everything back to normal.

helios210, 2020-07-06:
I had a similar problem on a OnePlus 7 Pro except my device showed "Mobile Network not Available" and with the module installed and active I could not access the "Sim & network" category in Settings. Disabling the module corrected the issue, I have uninstalled for now but will try an older version at the weekend to see if I have the same experience as TinyRaindrop.

VR-25, 2020-07-19:
This was fixed in July 3rd version.
Thanks

closed 2020-07-19 by VR-25

pannal, 2020-08-18:
Hey, I still have this issue with v2020.8.6 on my Mi 9T xiaomi.eu 12.0.4.0 stable.

kjanderson2, 2022-05-09:
> I have the same problem coming from xiaomi.eu firmware stable version 11 I flashed to the stable 12.0.1 noticed I couldn't make calls and then tried 12.0.2 but still having this problem. AFter [...]

Sorry, I am also experiencing this issue but am less well-versed on what this all means. When you sayd "after disabling the ACC Magisk module the problem is gone", how did you do that? How can I disable the ACC Magisk module? Thank you!

pannal, 2022-05-09:
> Sorry, I am also experiencing this issue but am less well-versed on what this all means. When you sayd "after disabling the ACC Magisk module the problem is gone", how did you do that? How can I [...]

If you don't know how to disable a module you probably shouldn't use magisk.

Anyhow: you can do that using magisk manager in the modules tab.