VR-25/acc #62: accd is constantly quitting itself?

whc2001, 2020-09-01:
I am using acc on OnePlus 8 Pro with the ROM of HydrogenOS 10.5.7.IN11 and rooted with Magisk. However it seem like that the acc daemon is constantly exiting, which I have to regularly open terminal and run acc to check and press 4 to restart if not running. Is that normal? If not, is there any workaround to keep it alive?

[acc-logs-OnePlus8Pro.tar.bz2.zip](https://github.com/VR-25/acc/files/5159268/acc-logs-OnePlus8Pro.tar.bz2.zip)

Hinfty, 2020-09-10:
I have the same issue with a OnePlus2 on LineageOS 17.1.
Will try to provide logs next time (usually happens every night)

Hinfty, 2020-09-11:
@VR-25 

This morning the same issue. I wanted to charge it to 40-60 during night and 90-95 in the morning. Have set a schedule on automate to run this at 6:00

`/dev/acca -s shutdown_capacity=0 resume_capacity=90 pause_capacity=95; /dev/acca --set --voltage 4200; /dev/acca -s cooldown_temp=35 max_temp=43 max_temp_pause=60; /dev/acca -s max_charging_current=1000; touch /sdcard/acc_profiles/profile-full;`

The touch is run correctly, but when I checked at 7:30, the phone was at 100% and acc was not running.

What's going on there?
(I also have logs, but can't seem to be able to upload them here?)

Hinfty, 2020-09-16:
Latest version seems to fix this for me. Acc keeps running through the night. Thanks!

whc2001, 2020-09-18:
The situation is the opposite for me. Just some time after posting this issue, accd seemed getting stable for a while, but when I updated it to the newest version, it automatically stops again and if I do not check the running status it will charge to 100%. I will watch further and provide log if it's still not working correctly.

VR-25, 2020-10-03:
The latest version should have this fixed.

closed 2020-10-03 by VR-25