VR-25/acc #63: Miui 12 Eu ROM Fastboot Issue

cramer52, 2020-09-09:
Hey, I tried to use acc with a Mi 10 Pro on Miui 12 Eu Rom and after the installation via magisk the device just reboots to fastboot. The only way to fix that issue is to reinstall the Rom and remove the module after the boot. That issue was already mentioned a couple of times in the acc telegram group, the xda thread and the xiaomi eu forum. Any idea how to solve this problem?

VR-25, 2020-09-09:
Apparently, this does not just relate to ACC alone.
I received reports from Migrator users as well - a completely different module.
Have you tried other Magisk modules?

cramer52, 2020-09-09:
A few but nothing really system relevant I guess. MagiskHide Props Config, JoyPixel, GpayFix, Miui GApps+ which worked all without that issue. Tell me if I can help fix that.

mgTerminator, 2020-09-19:
Hi!
Installed the latest version of the ACC my Poco F2 Pro, and no problems! 
Xiaomi.eu 20.9.17
MIUI 12.2 Android 11

yichiham, 2020-09-24:
I had the same fasboot issue running xiaomi eu 12.0.8 A10.
Now it's working like a charm with the latest v9.24 !

closed 2020-10-03 by VR-25