VR-25/acc #64: Not running on android stock rooted Marshmallow 6.0.1 Samsung On5 magisk 20.3

davidzam12, 2020-09-16:
Using the magisk repository I installed the last version, Previously I had no issues using Acc the acca apk and all to manage it. But now after reinstalling (I had to format my device) the last version under magisk 20.3 ( I can't upgrade to v. 20.4, It doesn't work either) It shows that:
```
accd
accd,
grep: /dev/.acc/ch-switches: No such file or directory (i) accd is not running
```
It doesn't run on any way or any command. `No such file or directory` is all the response I get, all the time, I reinstalled several times and the acca apk doesn't work either. any solution to this?

davidzam12, 2020-09-16:
Also by running the acc command
(i) accd is not running

then the menu shows

errors I get must be in the logs:

[config[1].txt](https://github.com/VR-25/acc/files/5229443/config.1.txt)
[install[1].log](https://github.com/VR-25/acc/files/5229448/install.1.log)
[power_supply-on5ltemtr[1].log](https://github.com/VR-25/acc/files/5229450/power_supply-on5ltemtr.1.log)
[power-supply-logger.sh[1].log](https://github.com/VR-25/acc/files/5229454/power-supply-logger.sh.1.log)

Also
[image]

Why the installation is there? I used magisk... follow every step correctly. Thank you.

VR-25, 2020-09-17:
Hi,
Try the following:
1. Reboot.
2. Run `accd`.
3. Run `acc -le`.
4. If `3` succeeds, share the output file. Else, compress and share `/sdcard/Download/acc/logs/`.

davidzam12, 2020-09-19:
I did as you indicated
and it shows me the acc config.txt file but nothing else I exited the terminal to start... the same errors... acc not found, file not found, folder not found... etc. It's not working.
[image]
LOGS:
[acc logs samsung on5.zip](https://github.com/VR-25/acc/files/5250489/acc.logs.samsung.on5.zip)

The AccA app manager it's not working either. As it did before...

davidzam12, 2020-09-19:
Any work around or solution for this?

davidzam12, 2020-09-19:
Here are the logs. AccA manager doesn't work either.

VR-25, 2020-10-03:
The latest version should have this fixed.

closed 2020-10-03 by VR-25