VR-25/acc #66: Does

dimscor, 2020-09-27:


dimscor, 2020-09-27:
After the last update of Lineage to 20082020 version accd doesn't works properly. When I start charging, it doesn't pauses as it should according to settings (60%). But I see, that when the charging reaches 60% it stops for 1 sec a few times, but then starts again.

VR-25, 2020-10-03:
The latest version should have this fixed.
Of yhw issues persists after upgrading, run `acc -t` to see which switches work, then `acc -ss` to manually set one.
For additional details, refer to readme > troubleshooting > charging switch.

closed 2020-10-03 by VR-25

dimscor, 2020-10-03:
The problem is that every switch works, but BattIdleMode=false. And last update didn't fix the issue. Charging doesn't stop permanently, but only for a few seconds.