VR-25/acc #68: ACC no longer working after either module or ROM update

jyscao, 2020-09-27:
* device: Xiaomi Mi Mix 2S (Polaris)
* ROM: LineageOS 17.1
* build number: lineage_polaris-userdebug 10 QQ3A.200805.001 7774ffa31a

I updated my ROM from an earlier build (dated 2020-08-28) to the current one (dated 2020-09-25), and also updated ACC from v2020.9.14 to v2020.9.24 a few days ago for both. I can't recall after which update it was that broke the functionality of ACC on my device, as I have only noticed it this morning when I left my phone charging overnight, and woke up to it charged to 100% which ofc shouldn't happen. I have only been using ACC on all its default settings, so it should only have charged up to 75% I believe.

I have attached my output logs from `#acc -le` (converted to gzip as GitHub doesn't support attaching bz2 archives). Just let me know if there's any other information/steps required.

[acc-logs-polaris.tar.gz](https://github.com/VR-25/acc/files/5288944/acc-logs-polaris.tar.gz)

VR-25, 2020-10-03:
The latest version should have this fixed.

closed 2020-10-03 by VR-25

jyscao, 2020-10-03:
Excellent thanks a lot!