VR-25/acc #69: ACC no longer broadcasting battery level in v2020.9.24

nowster, 2020-09-28:
On a Pixel 3a, installing acc v2020.9.24 freezes all battery level stats.

`acc -i` shows the correct battery level. However, nothing else on Android sees the correct battery stats.

Stopping acc daemon shows correct battery level in Android. (Android's battery service will be restarted.)

Reverting to v2020.9.14 fixes behaviour.

Is acc daemon correctly broadcasting battery level information?

I suspect the following change is the problem at line 214 in `acc/accd.sh`

```
-       if ! { ${capacity[4]} && [ $(cat $batt/capacity) -lt 2 ]; }; then
+       if ${capacity[4]} && [ $(cat $batt/capacity) -ge 2 ]; then
```

De Morgan's law is: _not (A and B) = not A or not B_

So the replacement should instead be:

```
    if ! ${capacity[4]} || [ $(cat $batt/capacity) -ge 2 ]; then
```

MAO04, 2020-09-29:
Same here on my pixel 3 XL.

VR-25, 2020-10-03:
The latest version fixes this.

closed 2020-10-03 by VR-25

CRCinAU, 2020-10-04:
@VR-25 Which version are you meaning?

I installed v2020.10.1 and still see this on a Pixel 3...

CRCinAU, 2020-10-04:
I can also confirm that using v2020.10.01 as a base, then making the change on line 216 to:
```
      if ! ${capacity[4]} || [ $(cat $batt/capacity) -ge 2 ]; then
```
causes the battery level within Android to work again.

pizzamx, 2020-10-04:
> @VR-25 Which version are you meaning?
> 
> I installed v2020.10.1 and still see this on a Pixel 3...
> 
> eg:
> [image]
> 
> ```
> root@blueline:/data/adb/modules/acc # ./acc.sh -v                                                                                                                                                                                            
> 
> v2020.10.1 (202010010)
> 
> root@blueline:/data/adb/modules/acc # 
> root@blueline:/data/adb/modules/acc # ./acc.sh -u                                                                                                                                                                                            
> 
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  3299  100  3299    0     0   4537      0 --:--:-- --:--:-- --:--:--  4544
> 
> 
> (i) No update available
> 
> root@blueline:/data/adb/modules/acc # 
> ```

confused too on Pixel 3

MAO04, 2020-10-06:
Still seeing this with v2020.10.1 too. On a pixel 3xl with Oct A11.