VR-25/acc #74: Doesn't respect lower limit.

Ph0rk0z, 2020-10-17:
On a nougat LG V20, using the magisk module and ACCA. Charge threshold set to 45-start -80 end.

When I reach 80% charging stops correctly. If I fall below 80% the phone will charge regardless. AccA reports -mah discharge but my charging switch says 1 and accubattery correctly shows positive MAH.

If I manually set the same switch to 0, charging of course stops. So looks like acc never sets the switch until battery "charge to" limit is reached.

I tested this right now.

Set to 46-> 57, battery was around 57
Charging stops, yay.
Battery falls to  56, still not charging, good.
Reboot phone. Charging switch is 1 again. Battery is 55%
Battery charges up again :(

VR-25, 2020-10-18:
Hi,

When battery level is between `resume_capacity` and `pause_capacity`, acc does nothing - i.e., it does not toggle charging switches.

Logically, the opposite would make no sense and would increase power consumption significantly.

The device is expected _not_ to toggle the charging switches back on under any circumstances.
It seems the opposite is true for LG V20, the kernel you're using or the particular charging switch set.

Most devices behave as expected - meaning, they don't reset the charging switches on plug/unplug.
The opposite is bad driver design (your case).

Another thing to note is that all changes made to control files (charging, current, voltage, etc.) do not persist across reboots.
This is true for all Unix/Linux-based systems and it's not an issue.

There may be a workaround for you.
Refer to readme > troubleshooting > charging switches.

Ph0rk0z, 2020-10-18:
I can plug the cable in as many times as I want. It's not doing that portion. ACC just never sets the switch at boot. The only time it does is when the charging threshold is reached. With a 10k mah battery sometimes you will reboot over the couple days it lasts.

VR-25, 2021-07-21:
Hi, I'm taking another look at this.
Ideally, acc should indeed deal with that "in-between" scenario too.
I'm working on that for the next release.
Already fixed the "negative current while charging" issue.
Thanks.

closed 2021-07-22 by VR-25