VR-25/acc #78: Not working on Pixel 5

leifalbor, 2020-11-15:
I'm guessing it's probably a kernel issue - but either way, here the logs to help troubleshoot.

Module flashes fine in the latest canary version of Magisk, but doesn't seem to actually do anything regardless of configuration (and charging switch) used.

[acc-logs-redfin.tar.bz2.zip](https://github.com/VR-25/acc/files/5543323/acc-logs-redfin.tar.bz2.zip)

TonyApuzzo, 2020-11-28:
Still not working with Magisk v21.1 and acc V2020.11.20.

No errors in Magisk log.

Stock build RD1A.200810.022.A4

TonyApuzzo, 2020-12-11:
Also not working on the December 2020 update with stock kernel, nor with ElementX 1.0.4 kernel.

NYCJames, 2021-01-12:
It also does not work on my Pixel 4a 5G Dec and Jan update

mrDoctorWho, 2021-07-19:
I can confirm it doesn't work on Pixel 5, but only after July update. It worked flawlessly before. It seems that all 4 charging switches are unavailable now. Is there anything we can do to make it work again?

VR-25, 2021-07-19:
I'll soon push an update with [new charging switches](https://github.com/VR-25/acc/blob/dev/acc/charging-switches.txt) and more.

leifalbor, 2021-07-19:
> I'll soon push an update with [new charging switches](https://github.com/VR-25/acc/blob/dev/acc/charging-switches.txt) and more.

Nice!  Hope you're doing well dude - I thought you must have abandoned the entire project!

mrDoctorWho, 2021-07-24:
> I'll soon push an update with [new charging switches](https://github.com/VR-25/acc/blob/dev/acc/charging-switches.txt) and more.

That would be great.

mrDoctorWho, 2021-07-29:
> I'll soon push an update with [new charging switches](https://github.com/VR-25/acc/blob/dev/acc/charging-switches.txt) and more.

Unfortunately, the update doesn't seem to bring any fix to this issue. I also noticed that the acc daemon crashes/stops working for some reason. When I occasionally run "acc" in terminal (or over adb) it says the daemon isn't running, even though I started it like a few minutes ago. Is there any particular logs you'd like to see?

I'm running latest stock on Pixel 5. I haven't had any issues with acc at all on my old phone (OG Pixel) and Pixel 5 (until the latest firmware update).

VR-25, 2021-07-29:
> > I'll soon push an update with [new charging switches](https://github.com/VR-25/acc/blob/dev/acc/charging-switches.txt) and more.
> 
> Unfortunately, the update doesn't seem to bring any fix to this issue. I also noticed that the acc daemon crashes/stops working for some reason. When I occasionally run "acc" in terminal (or over adb) it says the daemon isn't running, even though I started it like a few minutes ago. Is there any particular logs you'd like to see?
> 
> ~Charging switch test says none of them work, but the Acc App says that `dc/input_suspend` works, although I didn't notice it affecting charging.~
> UPD: The app says so for any switch, so I guess it's broken.
> 
> I'm running latest stock on Pixel 5. I haven't had any issues with acc at all on my old phone (OG Pixel) and Pixel 5 (until the latest firmware update). I also tried ElementalX kernel, but no luck with it either.

After the daemon stops, you can grab `/dev/.vr25/acc/accd-*.log`.
Alternatively, run `acc -l cat > /sdcard/accd.log`

regarding charging control, have you tried a regular adapter (a.k.a., slow charger)?

mrDoctorWho, 2021-07-29:
> After the daemon stops, you can grab `/dev/.vr25/acc/accd-*.log`.
> Alternatively, run `acc -l cat > /sdcard/accd.log`
Here you go: [accd.log](https://drop.helldev.net/35241596-f7af-49cc-b866-6f7fac245154/accd.log).

> regarding charging control, have you tried a regular adapter (a.k.a., slow charger)?
I usually use slow wireless charger. When I performed the tests from my last comment my phone was connected to a laptop via USB-A - USB-C cable.

VR-25, 2021-07-29:
> Here you go: [accd.log](https://drop.helldev.net/35241596-f7af-49cc-b866-6f7fac245154/accd.log). 

`exitCode=7` means all charging switches are failing - i.e., none of them responds when toggled.

> I usually use slow wireless charger. When I performed the tests from my last comment my phone was connected to a laptop via USB-A - USB-C cable.

I see.
This really has to do with the kernel.
Have you tried voltage and current control?
Speaking of current, `acc -ss` has presets of 0-500 mA.
You may want to try the lowest values and see if charging stops or at least slows down dramatically.
Monitor changes with `acc -i` or `acc -w1`.
Regarding voltage, it''s just a matter of testing whether the limit is honored.
`acc -sv 3920`: 3920 millivolts
`acc -sv -`: restore default

P.S., see if changing `switchDelay=2` to `switchDelay=5` in `/data/adb/modules/acc/misc-functions.sh` (line 379) solves the charging switch issue.
Restart accd after the change.
One-line: `sed -i '/switchDelay=2/s/2/5/' /data/adb/modules/acc/misc-functions.sh; accd`
Revert: `sed -i '/switchDelay=5/s/5/2/' /data/adb/modules/acc/misc-functions.sh; accd`

mrDoctorWho, 2021-07-29:
> Have you tried voltage and current control?

I tried just now and voltage limit seems to be working. As soon as I set the voltage limit to 3920, it stops charging and current becomes negative (battery voltage is over 4200mV now).

> Speaking of current, `acc -ss` has presets of 0-500 mA.
> You may want to try the lowest values and see if charging stops or at least slows down dramatically.

No luck with that, though. I tried `acc -s --current 10` and it kinda worked. But strange, though. Sometimes current goes up more than 800mA:
```
  CURRENT_NOW=784302
  CURRENT_NOW=805969
  CURRENT_NOW=-423889
  CURRENT_NOW=-246582
  CURRENT_NOW=-174865
  CURRENT_NOW=-195617
  CURRENT_NOW=-188293
  CURRENT_NOW=6408
  CURRENT_NOW=-53100
  CURRENT_NOW=810242
  CURRENT_NOW=-248413
  ````

I didn't notice anything like that with voltage limit.

> P.S., see if changing `switchDelay=2` to `switchDelay=5` in `/data/adb/modules/acc/misc-functions.sh` (line 379) solves the charging switch issue.
> Restart accd after the change.
> One-line: `sed -i [...]

That didn't work. accd says that switches don't work.

VR-25, 2021-07-29:
Interesting!
This suggests current works only partially.
Very small values are not honored.
Now, you may rely on voltage control as is - or use voltage control files as charging switches.
Share `/dev/.vr25/acc/ch-volt-ctrl-files`.

mrDoctorWho, 2021-07-29:
```
battery/constant_charge_voltage::v000::4300000
bms/voltage_max::v000::4450000
main/voltage_max::v000::4300000
sm7250_bms/constant_charge_voltage_max::v000::4300000
sm7250_bms/voltage_max::v000::4300000
```

VR-25, 2021-07-29:
[redfin_volt.txt](https://github.com/VR-25/acc/files/6899560/redfin_volt.txt)
[acc_latest_snapshot.zip](https://github.com/VR-25/acc/files/6899564/acc_latest_snapshot.zip)

There you go.
Test that list of potential switches - with acc -t /path/to/list (e.g., `acc -t /sdcard/Download/redfin_volt.txt`.
Use the latest acc zip attached.

mrDoctorWho, 2021-07-29:
```
redfin:/ # acc -t /sdcard/redfin_volt.txt

(!) Charger must be plugged to continue...
(i) Alright, this may take a minute or so...

(!) [battery/constant_charge_voltage 4300000 3300000] won't work

(!) [bms/voltage_max 4450000 3300000] won't work

(!) [main/voltage_max 4300000 3300000] won't work

(!) [sm7250_bms/constant_charge_voltage_max 4300000 3300000] won't work

(!) [sm7250_bms/voltage_max 4300000 3300000] won't work

redfin:/ #

```

mrDoctorWho, 2021-07-29:
Interestingly enough, it seems to be working now, but as strange as I mentioned before with current (goes up to 800+mA for 5 seconds for about once in 10-15 secs). It might be charging till 4200mV and then stops, then voltage drops to 4130mV and in a few seconds it starts charging again.

VR-25, 2021-07-29:
> Interestingly enough, it seems to be working now, but as strange as I mentioned before with current (goes up to 800+mA for 5 seconds for about once in 10-15 secs). It might be charging till 4200mV and then stops, then voltage drops to 4130mV and in a few seconds it starts charging again. Current battery level is 78%. Config is default.
> 
> Update: In about 10 minutes it charged to 79%.

That behavior is expected.
With a low voltage limit, the battery level will only go so far.
You can rely on this for now.

> ```
> redfin:/ # acc -t /sdcard/redfin_volt.txt
>
> (!) Charger must be plugged to continue...
> (i) Alright, this may take a minute or so...
>
> (!) [battery/constant_charge_voltage 4300000 3300000] won't [...]

This may be happening because the charging status is inconsistent - e.g., it may be fluctuating.
That's not an issue at all.
I just have to make acc _understand_ that logic.

mrDoctorWho, 2021-07-29:
> That behavior is expected.
> With a low voltage limit, the battery level will only go so far.
> You can rely on this for now.
> 
Well, thanks for the help! This is much better than nothing. But what is the right way to fix the problem with charging to the specified percent? Could this be fixed from your side or only from the kernel side?

VR-25, 2021-08-04:
> Well, thanks for the help! This is much better than nothing. But what is the right way to fix the problem with charging to the specified percent? Could this be fixed from your side or only from the kernel side?

The right way of fixing it is updating the kernel.
From my side, I can only provide workarounds.
Those may not satisfy all users.

Here's one more thing for you to try:
Upgrade to the latest snapshot and run `acc -s s=0 cw=true; accd`
Analyze charging control (pause, resume) for some time.
`cw` is short for `current_workaround`.
[acc_latest.zip](https://github.com/VR-25/acc/files/6929538/acc_latest.zip)

VR-25, 2021-08-05:
After reading a couple of posts on XDA, it became clear that this is a kernel issue.
Recent system updates have introduced it on Pixel devices.
(Google trying harder to enforce programmed obsolescence?)
Some users are suggesting switching to certain custom kernels, e.g., Proton, ElementalX (not the latest), etc..

Before going down that route, I suggest testing pairs of switches, like so:
`acc -t dc/input_suspend 0 1 sm7250_bms/charge_disable 0 1`

`acc -ss:` prinsts a list of known switches.
If a pair works, set it with `acc -s s="file1 on off file2 on off --"`, e.g., `acc -s s="dc/input_suspend 0 1 sm7250_bms/charge_disable 0 1 --"`

leifalbor, 2021-08-05:
Does ACCA support adding pairs of switches?  If so, how would this need to be formatted into the Add Charging Switch prompt?

VR-25, 2021-08-06:
> Does ACCA support adding pairs of switches? If so, how would this need to be formatted into the Add Charging Switch prompt?

file1 on off file2 on off --
e.g., dc/input_suspend 0 1 sm7250_bms/charge_disable 0 1 --

leifalbor, 2021-08-06:
No good I'm afraid.  Every combination I've tried, ACC reports back won't work.  Looks like if I want to use it, I'll have to go looking for a compatible kernel.

VR-25, 2021-08-08:
> No good I'm afraid. Every combination I've tried, ACC reports back won't work. Looks like if I want to use it, I'll have to go looking for a compatible kernel.

With accd stopped, have you tried writing to each control file manually and waiting in between (few secs) to see if charging stops?
Possibly, some switches work at least intermittently.
Again, it's a kernel issue, but we're looking for workarounds here.
It almost looks like OEMs have been introducing those kernel bugs intentionally.

gisthere, 2021-08-13:
Hi, this works for me:
`echo 1 > /sys/devices/platform/soc/c440000.qcom,spmi/spmi-0/spmi0-02/c440000.qcom,spmi:qcom,pm7250b@2:google,bms/power_supply/sm7250_bms/charge_disable`
It also changes _/sys/devices/platform/soc/c440000.qcom,spmi/spmi-0/spmi0-02/c440000.qcom,spmi:qcom,pm7250b@2:google,bms/power_supply/sm7250_bms/status_ to 'Not charging'
My build is: **RQ3A.210805.001.A1**

@VR-25 Could you explain please, why we need pair of switches instead of just one?

VR-25, 2021-08-14:
@gisthere, do `sm7250_bms/charge_disable`, `sm7250_bms/status` and the like exist in `/sys/class/power_supply/`?

Does `/sys/class/power_supply/battery/status` also change?

Does `acc -t sm7250_bms/charge_disable 0 1` work?
If not, try the full path.

Some devices (e.g., most MediaTek) require two switches - i.e., both have to be toggled for charging to stop/resume.
Some OnePlus phones also require two switches to enable idle mode.

gisthere, 2021-08-14:
@VR-25 thanks for the explanation!

> do sm7250_bms/charge_disable, sm7250_bms/status and the like exist in /sys/class/power_supply/?

Yes, they exist as symbolic links.

> Does /sys/class/power_supply/battery/status also change?

No

> Does acc -t sm7250_bms/charge_disable 0 1 work?
> If not, try the full path.

Seems, it doesn't work. But I noticed, that running 'acc -t ...' switches sm7250_bms/charge_disable from 1 to 0:

> redfin:/ # cat /sys/class/power_supply/sm7250_bms/charge_disable                                                                                                                                                                                                                 
> 0
> redfin:/ # echo 1 > /sys/class/power_supply/sm7250_bms/charge_disable                                                                                                                                                                                                            
> redfin:/ # cat /sys/class/power_supply/sm7250_bms/charge_disable                                                                                                                                                                                                                 
> 1
> redfin:/ # acc -t sm7250_bms/charge_disable 0 1                                                                                                                                                                                                                                  
> 
> (!) Charger must be plugged to continue...
> (i) Alright, this may take a minute or so...
> 
> (!) [sm7250_bms/charge_disable 0 1] won't work
> 
> redfin:/ # cat /sys/class/power_supply/sm7250_bms/charge_disable                                                                                                                                                                                                                 
> 0
>

VR-25, 2021-08-14:
The fact that `battery/status` does not change, while `sm7250_bms/status` does, reinforces the idea that the kernel is flawed.

Anyways, we're getting somewhere.

Have you confirmed that charging actually stops when status is `Discharging` or `Not charging`?
What does `acc -i curr` suggest?

Does `sm7250_bms/status` also change after toggling `battery/input_suspend`?

> I noticed, that running 'acc -t ...' switches sm7250_bms/charge_disable from 1 to 0:

That's because acc -t restores the default value, `0`.

gisthere, 2021-08-14:
> Have you confirmed that charging actually stops when sm7250_bms/status is Discharging or Not charging?
> What does acc -i curr suggest?

It seems like it really stops, because the phone becomes colder and percentage of the battery indicator (on the status bar) stops rising BUT:

> acc -i curr                                                                                                                                                                                                                             
> 
> Battery Service
>   Max charging current: 450000
> 
> Uevent
>   CONSTANT_CHARGE_CURRENT=2800000
>   CURRENT_AVG=4272
>   CURRENT_NOW=6408
>   INPUT_CURRENT_LIMITED=0
>   CURRENT_NOW=6.408 Amps

the value of CURRENT_NOW is strange. When I again put 0 to sm7250_bms/charge_disable, it becomes 0.289917 Amps.

> Does sm7250_bms/status also change after toggling battery/input_suspend?
I couldn't find battery/input_suspend, only dc/input_suspend.

> The fact that battery/status does not change, while sm7250_bms/status does, reinforces the idea that the kernel is flawed.
Hope, they will fix it Android 12

VR-25, 2021-08-14:
Ok, it seems charging does indeed stop.
The weird current values are just a unit conversion issue, no big deal (working on that too).

Here's the verdict so far...
The kernel is flawed.
It does not report the correct charging status in `/sys/class/power_supply/battery/status`.
While we wait for a kernel update, I'll implement a workaround - _invite `sm7250_bms/status` to the party_.

VR-25, 2021-08-14:
Install and test [this](https://github.com/VR-25/acc/files/6987085/acc_v2021.8.14.1_.202108141.zip) build.

gisthere, 2021-08-15:
Didn't help:
> 130|redfin:/ # acc -d
> (i) Manual mode
> 
> (i) Alright, this may take a minute or so...
> (i) /data/adb/vr25/acc-data/logs/acc-logs-redfin.tar.bz2
>                                                                                                                                                                                                                                                                                                                                                                                                                                                                     
> 130|redfin:/ # cat /sys/class/power_supply/sm7250_bms/status                                                                                                                                                                                                                     
> Charging
> 130|redfin:/ # acc -i stat 
> 
> Battery Service
>   status: 2
> 
> Uevent
>   CHARGE_CHARGER_STATE=50397205
>   STATUS=Charging

[acc-logs-redfin.zip](https://github.com/VR-25/acc/files/6988358/acc-logs-redfin.zip)

VR-25, 2021-08-17:
Here's another attempt to fixing the broken charging status.
This versions also lets you use voltage control files as charging switches.
That said, `acc -ss` has a new preset, 3600.
It limits voltage to 3600 in order to stop charging.
Essentially, whenever disable_charging() is called, voltage is limited to 3600 millivolts.
Keep mind that while limiting voltage may not always change the charging status, it still prevents charging past the set threshold.
That's essentially what I call _pseudo idle mode_.
The battery works as a power buffer.

[acc_v2021.8.17_(202108170).zip](https://github.com/VR-25/acc/files/7001322/acc_v2021.8.17_.202108170.zip)

VR-25, 2021-08-20:
[acc_v2021.8.20_(202108200).zip](https://github.com/VR-25/acc/files/7023359/acc_v2021.8.20_.202108200.zip)

giga-infinity, 2021-08-21:
> 
> 
> [acc_v2021.8.20_(202108200).zip](https://github.com/VR-25/acc/files/7023359/acc_v2021.8.20_.202108200.zip)

This fixed fixed my issue with latest versions not working on pixel5. Thank you VR-25!!

77Rudi, 2021-08-22:
On Pixel 4a it also runs since v2021.8.20 with 3600 preset. Thank you!!

VR-25, 2021-08-22:
It seems the official kernel fix isn't coming anytime soon.
We can't expect Google to care enough about this anyway.

I'm glad major progress has been made!
Yes, the workaround I implemented targets all devices with this issue.
Those who have been relying on a voltage setting may also try `acc -t`, since the regular charging switch approach should work as well.

closed 2021-08-30 by VR-25

mrDoctorWho, 2021-09-09:
It seems that it stopped working again after September update.

VR-25, 2021-09-10:
https://github.com/VR-25/acc/files/7138273/acc_v2021.9.9.1_.202109091.zip
Try that.

VR-25, 2021-09-10:
https://github.com/VR-25/acc/files/7143322/acc_v2021.9.10_.202109100.zip

VR-25, 2021-09-11:
https://github.com/VR-25/acc/files/7147378/acc_v2021.9.11_.202109110.zip
This build has a rollback feature.
To undo an upgrade, simply run `acc -b`.

If it stops working while in the background, share the output of `acc -l tail` after the occurrence - and before restarting the daemon.

mrDoctorWho, 2021-09-11:
```

VR-25, 2021-09-11:
Is this really today's build?

Anyway, you can bypass this error by manually setting the working switch with `acc -ss`.

mrDoctorWho, 2021-09-11:
Probably I did not restart the daemon before giving you the logs. Now it should be right:
```
251: off=1
252: chmod 0644 /sys/devices/platform/soc/soc:google,charger/charge_disable
252: >/sys/devices/platform/soc/soc:google,charger/charge_disable
252: eval echo '$off'
252: echo 1
252: return 1
252: exxit
89: exitCode=1
90: false
90: set +eux
```