VR-25/acc #79: ACC Daemon disables Battery Saver (& prevents enabling of battery saver mode)

smudgered, 2020-11-16:
The ACC Daemon disables Battery Saver & prevents enabling of battery saver mode, either via status bar or directly in settings.

Stopping the ACC Daemon removes this block, until phone is rebooted. Oreo AOSP - installation via ACCA front end.

(On tablet, also running Oreo, but in MiUI, this is not an issue as battery saver mode not available as an option, MiUI having its own implementation).

VR-25, 2020-11-17:
Weird...
I wonder whether this could this be related to `dumpsys`.
Does running `dumpsys battery reset` fix it - at least temporary? You have to try enabling battery saver mode immediately after running the command. If necessary, try this sequence multiple times.

Next, run `acc -le` and share the generated log archive.
It contains no personal/private information.

smudgered, 2020-11-17:
I thought I had upgraded acc by mistake when trying at enter the accd, command.
And guess what. It is now working! Confirmed working after reboot. 

Strangest thing is that the upgrade can't have completed properly as I still have the 202010150 version installed. I am having the devil of a job entering Ctrl-C btw using my terminal emulator but just found out long-press 😏

If it starts playing up I'll get back. Thanks for getting back.

smudgered, 2020-11-17:
For info

Starting acc wizard solves problem. Problem came back after attempting to set custom charging power controls > which led to reboot (system on phone couldn't accept the changes presumably) and blocked battery saver.

So on previous experience I simply started up the acc wizard in a terminal. Purely this alone released battery saver mode. Now ok when accd running 😁

My tablet accepts charging power controls.

(Tentative feature suggestion, an option f) in acc wizard to output acc - i command)

closed 2021-07-21 by VR-25