VR-25/acc #83: Samsung SM-T580 stops charging at 70% despite batt_slate_mode charging switch

andiwirs, 2021-01-12:
Hello, first of all thanks a lot for ACC. However i have a problem with my Samsung SM-T580 Tablet. 
Since it always stopped at 70% i changed the charging switch to "batt_slate_mode" according to the troubleshooting section. Furthermore i disabled that the switch can be changed automatically. However it still stops at 70%.

Any hints or suggestions what to try? If it is a bug, how can i help you find a fix for it?

Thanks in advance!

andiwirs, 2021-01-18:
Ok ich guess it is an ACC bug. Uninstalled it and tried the app "battery charge limit" and this works fine. This also uses the control file "batt_slate_mode" to stop charging

andiwirs, 2021-03-17:
Observed it for some time now and it seem fixed with release **v2021.2.25 (202102250)**
Guess it has to do with "Enforce control files mode 0644, so that other processes cannot reset/override changes made by acc."

closed 2021-03-17 by andiwirs

Nemris, 2021-04-04:
@VR-25 I can reproduce this issue with acc v2021.2.25 after updating to stock rooted Android 11 on a SM-A405FN, hence One UI 3.1.
No matter what, charging stops at 70% and doesn't appear to drop for as long as the device is connected to a charger. Using `battery/batt_slate_mode 0 1` doesn't help.

Might be worth mentioning this issue didn't manifest under stock Android 10.

In case you need any logs, just say the word.