VR-25/acc #89: `--` not respected in charging switch options

RellikerV2, 2021-03-05:
For some reason appending `--` to the charging switches does not work to prevent accd from trying all the switches again if it thinks the switch didn't work, contrary to documentation. The switch listed below works fine because with acc fully killed setting it to 1 manually works fine to prevent charging when plugged.

Device: OnePlus 8 Pro (instantnoodlep) 
Android: LOS 11

```
instantnoodlep:/sys/class/power_supply/battery # acc --version

v2021.2.25 (202102250)

instantnoodlep:/sys/class/power_supply/battery # grep ^chargingSwitch /sdcard/Documents/vr25/acc/config.txt
chargingSwitch=()
instantnoodlep:/sys/class/power_supply/battery # acc -s s="battery/op_disable_charge 0 1 --"

instantnoodlep:/sys/class/power_supply/battery # grep ^chargingSwitch /sdcard/Documents/vr25/acc/config.txt
chargingSwitch=(battery/op_disable_charge 0 1 --)

<replug charger and wait ~20s>
instantnoodlep:/sys/class/power_supply/battery # grep ^chargingSwitch /sdcard/Documents/vr25/acc/config.txt
chargingSwitch=()
```

I also tried `battery/op_disable_charge 0::1 --` with the same issues. The syntax in the docs is also inconsistent, sometimes using spaces between the sysfs file and sometimes using double colons... what is the accepted way of using them now?

mite1200, 2021-03-08:
Having the exact same issue on a Pixel 5. It's very annoying. Anyone know a fix?

VR-25, 2021-07-10:
Working on updates.

"::" is used as a _space placeholder_ for MediaTek specific charging switches.
Having spaces there makes line parsing way more complex than it should be.

Example:
chargingSwitch=(/proc/mtk_battery_cmd/current_cmd **0::0 0::1** /proc/mtk_battery_cmd/en_power_path 1 0")

0::0 0::1 could have been written as "0 0" "0 1".
However, isolating the quoted characters during parsing is just as inefficient as processing them unquoted, **0 0 0 1**.

I'll add this information to the documentation.

VR-25, 2021-07-21:
The upcoming release has this fixed.