VR-25/acc #90: Acc disable in magisk continue starting

remdu92, 2021-03-12:
Hi guys,

I've installed acc magisk module and acca app to manage it.
I have disabled the acc module in magisk but it's still starting.
I've found with htop that the deamon start with init.d second stage.

After 1hour of research, I can't find where the boot flag is setup ...
Module is disabled so I can not see it in magisk log.

Does someone know why it keep starting ? Related to acca app ?

Thanks in advance.

Ps : last magisk and acc module on a galaxy s20 4g

bege10, 2021-06-01:
Same issue on Fairphone 3, /e/OS (based on FairphoneOS, LineageOS for microG, Android 10), Magisk 23001, ACC 2021.2.25, AccA 1.0.29.

VR-25, 2021-07-10:
AccA does not honor Magisk's _module disable_ flag.
Thus, the app is launching `accd` on `boot_completed`.
I'll make acc daemon quit if that flag is set.
Thanks.

Rihcus, 2021-07-10:
Not sure if this belongs here, does uninstalling acc via magisk instead of from say termux or shell fully remove acc and any modifications. Additionally does acc modify any partitions outside of boot.img of the source rom (ie If I where to reinstall lineage os would modifications from acc still persist).

VR-25, 2021-07-11:
> Not sure if this belongs here, does uninstalling acc via magisk instead of from say termux or shell fully remove acc and any modifications. Additionally does acc modify any partitions outside of [...]

Yes, uninstalling acc via magisk fully removes it too.
Magisk executes acc's uninstall.sh (cleanup script) upon uninstall.
ACC is always systemless - meaning, it never touches any partition besides data.
Modifications made by it to control charging stay in RAM only - a reboot/shutdown resets them all.

VR-25, 2021-07-21:
The upcoming release has this fixed.

closed 2021-07-21 by VR-25