VR-25/acc #91: [Request]restore max charge voltage regulation up to 4300mV

MeloMCV, 2021-03-13:
After a couple updates, and since android 11 the max charge voltage is modified up to 4200 mV and for me is too low, i had set it to 4300 for good between autonomy and battery life. If possible please modify it. Good work really!

VR-25, 2021-03-18:
Hi,

Thanks for the suggestion.
While waiting for it to be implemented, you can override the 4200 mV limit as follows:
`# acc -sv 4200 && acc -c sed -i s/:4200/:4300/g && accd`

P.S., The limit is 4200 mV because older batteries don't go over that.
Thus, it's a safety measure.
Although, acc doesn't mind if you edit the config file manually to override the limit - and `acc -c sed -i s/:4200/:4300/g` is a way of doing that.

MeloMCV, 2021-03-18:
I tried it with terminal but doesn't work, still 4200mV limit.. Anyway, no problem if can't adjust it right now. If it is possible in some newer version you can put in natively, no problem. Thank you anyway for your courtesy and presence. You made a really good job with acc and i'm very happy

VR-25, 2021-07-21:
The upcoming release is bringing this change.

closed 2021-07-21 by VR-25