VR-25/acc #92: Warp charging on OnePlus6T doesn't follow control file

Fireeyeeian, 2021-03-14:
Notes: I got ACC though Magisk and also got the app gui but also use the terminal for logs and setting control file

So Acc seems to works as expected when charging without fast charge (warp charge or whatever it's called on the op6T) but when I plug my phone into the fast charge brick the device stops and then starts charging again and then repeats I read in the FAQ that some devices do this and tried what it said to do but I have had no luck. I have also noticed that the phone no longer shows the fast charge animation on the lock screen anymore (not a huge deal but odd).
I have attached a log of the situation.

[acc-logs-OnePlus6T.zip](https://github.com/VR-25/acc/files/6136540/acc-logs-OnePlus6T.zip)

 A side issue is that I cannot stop acc at all.. I've tried using the gui app (when I use the app nothing changes.. it stays green and says its running) and also tried through the terminal by typing `acc stop`, `accd stop`, and  `accd. Stop` and It says it stopped but the function is still working and the app says it's running still.
Any help is appreciated and if I'm doing something stupid let me know. 
-Thanks in advance

VR-25, 2021-07-09:
Hi there,
I've been out for some time.
I'm working on updates now.
We've been having issues with _wicked_ fast charging for a while.
I'm encouraging people to use regular power adapters as this issue undergoes further investigation.
Thanks for the detailed information and logs.
Ensure you're using the latest versions of acc and acca.

Fireeyeeian, 2021-07-10:
Thanks for the update, and yes I have been using slow chargers for the time
being as it's not too big of a deal.

VR-25, 2021-07-22:
If using another charging switch does not solve this, I'm afraid the only solution is a regular power adapter.
I added info regarding this issue to the documentation (upcoming release).

closed 2021-07-22 by VR-25