VR-25/acc #93: Current control not working on 6T

droid-jd, 2021-03-22:
I'm running LOS 17.1 on a oneplus 6T with MCD kernel. Got months I've been able to set the current restrictions to 500ma in acca with no problem. Recently, the current control doesn't seem to be working. I've tried setting different values from 250-700 in both acca and acc via terminal emulator. Capacity control and idle mode each work fine.

Log output below screenshots.

Thanks for any help!

[image]
[image]
[image]

[su -c acc -le output](https://www.dropbox.com/s/qo9zkk8anmh76ax/acc-logs-OnePlus6T.tar.bz2?dl=0)

nphantasm, 2021-04-14:
I can confirm this issue on Xiaomi A2 Lite running LineageOS 18.1 with stock kernel. Capacity control still works fine.
Using older versions doesn't have any effect.

[acc-logs](https://drive.google.com/file/d/1RI5j0bG9h9c9N-E9F6j21UnYBAPsS0s_/view?usp=sharing)

VR-25, 2021-07-09:
The latest version should have this fixed.
The changelog states: reapply current settings (e.g., `acc -sc 1800`) after upgrading AND rebooting.
Let me know if the issue persists after that.

closed 2021-07-09 by VR-25