VR-25/acc #94: accd consume too much battery itself

ioctl-user, 2021-04-03:
Phone with accd discharge 1.5 time faster at the night. This can be proven using the Battery Historian PC tool.

It seems, because it's internal cycle is 10 seconds, so it wakes up and consume battery.

michael-mml, 2021-04-15:
I believe it's holding a wakelock on the CPU, my device is idling at 300 MHz rather than entering deep sleep

ioctl-user, 2021-04-16:
Also it can be because accd has 10 seconds status check cycle. So, it should wake up CPU each 10 seconds.

VR-25, 2021-07-09:
This may be related to a rogue charging switch.
Other users -myself included - would be facing the same issue.
Refer to https://github.com/vr-25/acc#charging-switch .
If the issue persists after following that guide, share logs (`acc -le`).

ioctl-user, 2021-07-13:
Now I am running other firmware with other apps set. It seems, there is no difference in power consumption between these cases:

*) Accd not running.
*) Accd running with switch cycle.
*) Accd running without switch cycle.

I am not running Acc directly, but using AccA app to manage it.

closed 2021-07-13 by ioctl-user