VR-25/acc #95: ''Bad address" error

OraBB1t, 2021-04-08:
This started happening recently. I think(not sure) it started when I switched to lineageOS. I tried re-installing acc with no luck. ACC still works with the default config though. I just can't configure it.

Device - OP5
Root - Magisk
ROM - lineageOS 17.1

```
~ $ sudo acc

/system/bin/acc: /data/adb/vr25/acc/misc-functions.sh[375]: /dev/.busybox/readlink: Bad address
/system/bin/acc: /data/adb/vr25/acc/misc-functions.sh[379]: /dev/.busybox/pgrep: Bad address
/system/bin/acc: /data/adb/vr25/acc/misc-functions.sh[382]: /dev/.busybox/grep: Bad address
/system/bin/acc[159]: /dev/.busybox/touch: Bad address
/system/bin/acc[159]: /dev/.busybox/cut: Bad address
/system/bin/acc[159]: /dev/.busybox/du: Bad address
/system/bin/acc[159]: [: 2: unexpected operator/operand
/system/bin/acc[159]: /dev/.busybox/date: Bad address
/system/bin/acc[162]: /dev/.busybox/sed: Bad address
/system/bin/acc[163]: /dev/.busybox/sed: Bad address
/system/bin/acc[516]: /dev/.busybox/more: Bad address

/system/bin/acc: /dev/.busybox/mkdir: Bad address
```

OraBB1t, 2021-04-10:
I was using termux to configure acc. Just tried another terminal emulator and it worked fine. Problem seems to be with termux, not with acc. Closing.

closed 2021-04-10 by OraBB1t

iwansal64, 2025-02-05:
Which other terminal emulator did you ended up?