VR-25/acc #96: ACCD Quits Unexpectedly

Nemris, 2021-04-11:
Device: `SM-A405FN`
OS: `One UI 3.1 (Android 11)`
Magisk Version: `22.0`, `22.1`

On my device, `accd` appears to stop unexpectedly at random times after the OS upgrade to Android 11. The issue is more likely to trigger after the phone has been on since more than twenty hours.

If logs are needed, I'll upload them.

dreamflasher, 2021-04-11:
I can confirm this, currently on Android 11 acc doesn't work due to this. For me accd constantly stops running, even after a few minutes.

dreamflasher, 2021-04-18:
I just downgraded to Android 10 and I observe the same behaviour. Makes me wonder if it's a problem with the latest acc version, I'll try previous versions there, too.

dreamflasher, 2021-04-18:
``v2020.11.20 (202011200)**``also not working :(

dreamflasher, 2021-04-18:
Seems this has been an issue previously https://github.com/VR-25/acc/issues/62

dreamflasher, 2021-05-04:
ping @VR-25 is acc working for you on Android 10 or 11?

kuzi-moto, 2021-06-16:
I would like to confirm with a Pixel 3 on Android 11. Just installed ACC and can't get it to stay running for longer than 10 seconds.

A status update could be useful, at least just to know whether or not to expect any updates soon.

VR-25, 2021-07-09:
Hello,
Could one of you share a log archive generated by `acc -le`?
Note that in order to get relevant information, the logs must be grabbed as soon as possible after the daemon stops unexpectedly.
Thanks.
I'm back and working on updates.

Nemris, 2021-07-16:
[acc-logs-a40.tar.bz2.zip](https://github.com/VR-25/acc/files/6833660/acc-logs-a40.tar.bz2.zip)
@VR-25 Here's a sample from today - remove the phony `.zip` extension, though, as GitHub dislikes `.tar.bz2` files.

VR-25, 2021-07-21:
`77: . /sdcard/Documents/vr25/acc/config.txt
/data/adb/vr25/acc/accd.sh[464]: .: /sdcard/Documents/vr25/acc/config.txt: No such file or directory
77: exxit
35: exitCode=1 `

This suggests that Android may be suspending the emulated storage when the system is idle.
Doe anyone have any other thoughts?

Here's a pro tip, by the way:
Run `acc -l tail` to quickly find why acc daemon stopped.
This prints the last 10 lines of the daemon log file.

Nemris, 2021-07-21:
I'll do it as soon as I notice the daemon stopping again.

tbertels, 2021-07-22:
> 
> 
> ```
> 77: . /sdcard/Documents/vr25/acc/config.txt
> /data/adb/vr25/acc/accd.sh[464]: .: /sdcard/Documents/vr25/acc/config.txt: No such file or directory
> 77: exxit
> 35: exitCode=1
> ```
> 
> This suggests that Android may be suspending the emulated storage when the system is idle.
> Does anyone have any other thoughts?

Maybe AAC config was restored with the Migrate app but the file /sdcard/Documents/vr25/acc/config.txt doesn't exist anymore. Even without Migrate, maybe the sdcard path has changed or the file isn't there anymore.
I've had a similar problem after restoring with Migrate on Android 11:
`81: eval ''
0: cat battery/temp
/data/adb/vr25/acc/accd.sh[464]: temperature[3]: parameter not set
84: exxit
35: exitCode=1
36: set +eux`
The daemon was also stopping after a few seconds.
I just disabled battery temperature check and saved the config with AccA and it worked.  Afterward, I reenabled the temperature check.

@Nemris Have you tried changing a setting then saving the config in AccA?

Nemris, 2021-07-22:
@tbertels I've been using `acc` directly after AccA gave issues some time ago, so I can't confirm that. Either way, `/sdcard/Documents/vr25/acc/config.txt` _appears_ to exist while I look at the phone, so I wonder whether the culprit might be an overzealous Samsung optimization behind the scenes.

Still, if the issue depended on FUSE itself - as is the case with the `/sdcard`, `/storage/emulated/0` and such mountpoints - then maybe using the raw `/data/media/0` could bypass the issue.
It's not at all classy and it means we would need to manually handle ownership and permissions - such as files and directories belonging to `media_rw` and having respectively `664` and `775` permissions - but it should be doable in case `acc` runs as root.

Can you test that, @VR-25?

VR-25, 2021-07-23:
> @tbertels I've been using `acc` directly after AccA gave issues some time ago, so I can't confirm that. Either way, `/sdcard/Documents/vr25/acc/config.txt` _appears_ to exist while I look at the phone, so I wonder whether the culprit might be an overzealous Samsung optimization behind the scenes.
> 
> Still, if the issue depended on FUSE itself - as is the case with the `/sdcard`, `/storage/emulated/0` and such mountpoints - then maybe using the raw `/data/media/0` could bypass the issue.
> It's not at all classy and it means we would need to manually handle ownership and permissions - such as files and directories belonging to `media_rw` and having respectively `664` and `775` permissions - but it should be doable in case `acc` runs as root.
> 
> Edit: actually, assuming using `/data/media/0` works, only reading the `config.txt` should require us to act as root. Since the FUSE mountpoints should be available at all times when the phone is unlocked, there would be no need to _write_ to `/data/media/0` at all.
> 
> Can you test that, @VR-25?
Working on it.
We're on the same page.
Great!

VR-25, 2021-07-29:
The new data path is `/data/adb/vr25/acc-data/`.
This is even better than `/data/media/0`, as it's more secure.
Public paths allow malicious app to easily run commands as root, using acc config file.

closed 2021-08-05 by VR-25

Nemris, 2021-08-05:
The program doesn't stop in the background any longer, for the record.