VR-25/acc #97: accd dies after plugging usb cable

GiulioBen, 2021-04-15:
Doogee S96 Pro

the daemon is stopper when i plug the cable. 
If i restart it, it works fine until i disconnect and reconnect the cable. 

https://user-images.githubusercontent.com/38356385/114903869-83f15600-9e17-11eb-9db3-b06c25e8d6d4.mp4

the log
https://www.dropbox.com/s/vq5oiuat76q7e2w/acc-logs-S96Pro.tar.bz2?dl=1

thank you

530f6caa, 2021-04-25:
It seems this occurs because `sync_capacity` will exit with an error when called from `wait_plug` because `isCharging` is not set in this context. (`sync_capacity` works when called from `is_charging`.)

GiulioBen, 2021-04-25:
how should i fix it?
I'm using it at almost stock configuration.

If i restart accd, it works until the next time i disconnect and reconnect
the cable

530f6caa, 2021-04-27:
I have made the following change:

```
--- a/acc/misc-functions.sh
+++ b/acc/misc-functions.sh
@@ -307,6 +307,7 @@
 
 
 wait_plug() {
+  isCharging=false
   $isAccd || {
     echo "(i) ghostCharging=true"
     print_wait_plug
```

I don't know if this is the 'right' way to fix this, but with light testing over the last day, `accd` doesn't crash and seems to be working as expected.

GiulioBen, 2021-04-27:
At the moment I installed acc as masgisk module. 
How should I proceed to install the new version with your edit?

Unistall the magisk version and install a new one built from github? 
thank you.

530f6caa, 2021-04-29:
The above patch hasn't been merged into the GitHub version, however you could apply it by modifying the file `/data/adb/vr25/acc/misc-functions.sh` directly on your device (e.g., via root shell).

GiulioBen, 2021-04-29:
thank you. i made the change, I'll test it and give u a feedback.

GiulioBen, 2021-05-01:
WORKED LIKE A CHARM!

T H A N K...... Y O U...... B O S S!!

530f6caa, 2021-05-03:
Glad to hear the fix worked for you. Can you please re-open this issue so that it remains open until this fix (or a better fix) is integrated into the main codebase?

VR-25, 2021-07-09:
@530f6caa, thanks for saving me some time!

closed 2021-07-09 by VR-25