VR-25/acc #99: Motor G4 Play XT603 with LineageOS 17 weird battery issues

BrunoDSL, 2021-04-22:
This is a problem I've been experiencing for quite a while and I wasn't sure if I shoould report it, as I was told this is most likely a battery issue.

Since I installed LineageOS 17, my battery was never the same. It now shutdowns at random percentages from 38% to 25%, sometimes 20%, rarely 15%. After installing acc, I can sometimes use the full capacity of the battery...But only every other charge. After I have one full battery charge usage, the next charge will have the issue. Is this a battery issue?

BrunoDSL, 2021-04-22:
I don't know if the log file will have relevant information, as running `su -c acc --log --export` made the phone reset the first time.

https://drive.google.com/file/d/1Oi5Rjo5NDj-GnlBBk0dr7GcnrzGkdlS0/view?usp=drivesdk

VR-25, 2021-05-01:
The phone I test stuff on is also a Moto G4 Play (XT607).
This is indeed a battery problem.
More specifically, it's due to sudden battery voltage drop.
I adapted a battery from Samsung Galaxy S4 and it's been serving me well ever since.

BrunoDSL, 2021-05-01:
> The phone I test stuff on is also a Moto G4 Play (XT607).
> This is indeed a battery problem.
> More specifically, it's due to sudden battery voltage drop.
> I adapted a battery from Samsung Galaxy S4 and it's been serving me well ever since.

I noticed that the S4 battery has similar capacity (2600 mAh), the fit seems similar, but the voltage required on it is higher (4.3v. The GK40 battery requires 3.8v). How would that work?

VR-25, 2021-07-06:
Both batteries charge to 4.4 Volts max and have the same [nominal](https://www.large.net/news/8qu43pq.html) voltage, 3.8 Volts.

closed 2021-07-09 by VR-25