VR-25/acc #103: Installation failed on LineageOS 14.1

godfuture, 2021-07-25:
I have been using acc for a long time, but it seems since a year I cannot install acc anymore. When installing, it always tells "installation failed!". 

Is acc not compatibel with Android 7 anymore? If yes, which version should Android 7 users install?

VR-25, 2021-07-25:
ACC should install on Android 4.0+ (or even lower, if mksh is manually upgraded).

I would like to see logs.
Try installing the latest version and share init.log and install.log from `/sdcard/Documents/vr25/acc/logs/`.

godfuture, 2021-07-25:
I cannot open /sdcard/Documents/vr25 no matter what app I use. Owner is root and group is everybody. Both have read permission.
What could I do?

VR-25, 2021-07-25:
Try `/data/media/0/Documents/vr25/`.

godfuture, 2021-07-26:
/data/media/0/Documents/vr25/ worked for me, thanks.
[Acc.zip](https://github.com/VR-25/acc/files/6881613/Acc.zip)

VR-25, 2021-07-29:
Have you tried the [latest](https://github.com/vr25/acc/releases/latest) release?

godfuture, 2021-08-05:
Latest release work just fine for me. Now also the AccA app works. Thanks!

closed 2021-08-05 by godfuture