VR-25/acc #107: BAG: Reboot my phone after send: "--Set --Current -"

nialone, 2021-08-05:
If open the terminal and send the "--set --current -" root command, my phone reboots. )
Where can I see logs? What could be?

I temporarily disabled this command in my ACCA build. Works good.

nialone, 2021-08-05:
Sorry, this is a machine translation.
I write the [problem](https://github.com/VR-25/acc/issues/107) in the ACC branch.
Thank you.

VR-25, 2021-08-05:
This means writing to some current control files causes a kernel panic - which then triggers a reboot.
It's a kernel/SoC specific issue.
The offending control file has to be blacklisted.
Share the log archive generated by `acc -le`.

nialone, 2021-08-05:
Log after reboot 
[acc-logs-S99.zip](https://github.com/VR-25/acc/files/6940818/acc-logs-S99.zip)

VR-25, 2021-08-06:
> Log after reboot
> [acc-logs-S99.zip](https://github.com/VR-25/acc/files/6940818/acc-logs-S99.zip)
> 
> The offending control file has to be blacklisted. How to do it? Variants?
> May be add "not use -set -current" the option or check in Acc\AccA ?

I do the blacklisting after looking at the list of control files from the log archive.
The problem is that your list is actually empty - which does not make sense if the system is rebooting.
See if `/dev/.vr25/acc/ch-curr-ctrl-files` is also an empty file.

nialone, 2021-08-06:
Yes, /dev/.vr25/acc/ch-curr-ctrl-files is empty. Why?

VR-25, 2021-08-06:
It means acc did not find any known charging current control file in your system.
That said, the possibility of a particular control file triggering reboots is now out of question.
This issue just moved to the unknown and weird territory.
We're back to ground zero - figure out what else could be causing unwanted reboots.

nialone, 2021-08-07:
Please see another phone with the same problem without rebooting, only freeze in AccA.
File сh-surr-ctrl is empty, ch-volt-ctrl file is not empty: "Battery/voltage::v::4400"
[acc-logs-HWSTF.zip](https://github.com/VR-25/acc/files/6948775/acc-logs-HWSTF.zip)

nialone, 2021-08-07:
upd2:  The second phone:
I connected charging to the phone, all files are not empty.
I tried to apply a profile via ACCA, the phone works.
I do not understand the logic of the work.

On the first phone, the current file is still empty.

VR-25, 2021-08-08:
> upd2: The second phone:
> I connected charging to the phone, all files are not empty.
> I tried to apply a profile via ACCA, the phone works.
> I do not understand the logic of the work.

No issues there...
What happens is that some current control files have a value of zero while the device is unplugged.
Thus, **once after each boot**, in order to store the real default current values for a later restore, acc waits until the charger is plugged.

> On the first phone, the current file is still empty.

I'll build a debug zip to log what happens just before the reboot is triggered.

VR-25, 2021-08-10:
Does `acc -t` also trigger a reboot?

nialone, 2021-08-11:
YES. I see in the terminal: "You need to connect the charger ..." 
We are waiting a few seconds and the phone reboots

VR-25, 2021-08-11:
Try the attached build (`acc -t`).
[acc_v2021.8.11_(202108110).zip](https://github.com/VR-25/acc/files/6971732/acc_v2021.8.11_.202108110.zip)

nialone, 2021-08-12:
I entered: "acc -x -t", - rebooting..
[acc-S99.log](https://github.com/VR-25/acc/files/6974376/acc-S99.log)

VR-25, 2021-08-12:
Ok, this is not an acc bug.
It's a kernel issue.

Which of the following commands trigger the reboot?
`echo "0 0" > /proc/mtk_battery_cmd/current_cmd`
`echo 1 > /proc/mtk_battery_cmd/en_power_path`

nialone, 2021-08-12:
Nothing is happening. it's all good.

nialone, 2021-08-12:
> Ok, this is not an acc bug.
> It's a kernel issue.
> 
> Which of the following commands trigger the reboot?
> `echo "0 0" > /proc/mtk_battery_cmd/current_cmd`
> `echo 1 > /proc/mtk_battery_cmd/en_power_path`

Nothing is happening. it's all good.

Ok. Command: "accd-x-D stop", also reboots. This is rare.
+ The log file is not created..

VR-25, 2021-08-13:
`accd -x -D stop` syntax is invalid.
It restarts the daemon, just like plain `accd`.

Let's try a different route:
1. Force stop the daemon with the `pkill -9 -f accd` command.
2. Run `acc -x -t,`acc -x -e`,`acc -x -d` and grab the log before each subsequent command - just to ensure the file is not overwritten.

Previous logs suggest that the system reboots just before accd stops.
Thus, force stopping it will bypass that, so we can see whether the 3 command above also trigger reboots.

nialone, 2021-08-14:
Grab?. I have to save the log 3 separately or?
1. I used PKILL before each command.
2. The report changed every time.
The phone is not rebooted.

???))

[image]

Pkill works strange, sometimes he can also cause a reboot.

VR-25, 2021-08-14:
That's very strange.
The charging switches are not triggering it either.
Try the commands `flock 0` and `cmd battery reset`.

nialone, 2021-08-14:
I entered it, there is no effect.

VR-25, 2021-08-14:
[acc_v2021.8.14.1_(202108141).zip](https://github.com/VR-25/acc/files/6987085/acc_v2021.8.14.1_.202108141.zip)
Install the attached build and test the commands that trigger a reboot.

nialone, 2021-08-14:
See this file. Line 2913. Will this help? I entered: accd.

[cat proclast_kmsg .txt](https://github.com/VR-25/acc/files/6987260/cat.proclast_kmsg.txt)

VR-25, 2021-08-15:
Still no clue on the actual cause of the reboots.
Run `pkill -9 -f accd`, install the new zip, run `accd.` (system will reboot) and share /sdcard/accd-*.log.

[image]

nialone, 2021-08-15:
The developer version is already installed.
The command "accd." does not create a report.
I realized the difference, reboot 100%, if the charge is connected.
I entered: "accd.", "acc -x -d", "acc -x -e" - all reboots.
If you disable charging, how lucky, 50/50. 
Inside the report, all problemary end on "timeout 10 Flock 0" 
[acc-S99-2.log](https://github.com/VR-25/acc/files/6987794/acc-S99-2.log)

VR-25, 2021-08-15:
> The developer version is already installed.

It's not the same.
I made changes, but did not update version numbers.
That zip generates even more verbose and redirects it to `/sdcard/accd-*.log`.

nialone, 2021-08-15:
Reinstalled on the last
[accd-S99.log](https://github.com/VR-25/acc/files/6988028/accd-S99.log)

nialone, 2021-08-15:
[accd-S99.log](https://github.com/VR-25/acc/files/6988083/accd-S99.log)
[acc-S99.log](https://github.com/VR-25/acc/files/6988084/acc-S99.log)

VR-25, 2021-08-17:
> Reinstalled on the last. Entered: "acc -s c -". The others is without failure.
> [accd-S99.log](https://github.com/VR-25/acc/files/6988083/accd-S99.log)
> [acc-S99.log](https://github.com/VR-25/acc/files/6988084/acc-S99.log)

What about `accd.`, `acc -x -d` and `acc -x -t`?
Only share logs if a reboot is triggered.
So far, this is the weirdest MediaTek related issue I came across.

nialone, 2021-08-17:
> What about `accd.`, `acc -x -d` and `acc -x -t`?

`accd.` - The "accd-*" file is not created when rebooting
`acc -x -d` - The "acc-*" file is not recorded when rebooting 
`acc -x -t` - [acc-S99-t.log](https://github.com/VR-25/acc/files/7002880/acc-S99-t.log) - After rebooting

Power cable connected to the phone .

VR-25, 2021-08-20:
[acc_v2021.8.20_(202108200).zip](https://github.com/VR-25/acc/files/7023359/acc_v2021.8.20_.202108200.zip)
This version has extra safety mechanisms to ensure defaults are always restored on daemon stop/restart.
This means neither `timeout 10 flock 0` nor `kill -KILL` are used.
Thus, the daemon will always be stopped _gracefully_.
A potential drawback of this is that the charger must be plugged if the install, upgrade, stop or restart processes seem to take too long.
There's yet no way to mitigate such negative side effect.

nialone, 2021-08-20:
Thank you very much for your help.
acc_v2021.8.20_(202108200).zip

`accd,` - normal
`acc -v` - normal
`accd -v` - rebooting
`accd.` - rebooting
`accd -x -d` - rebooting [acc-S99-d.log](https://github.com/VR-25/acc/files/7023651/acc-S99-d.log)
`accd -x -e` - rebooting, log is emty
`accd -t` - rebooting [acc-S99-t.log](https://github.com/VR-25/acc/files/7023654/acc-S99-t.log)

ps: Power cable connected to the phone . As if nothing has changed ))

VR-25, 2021-08-21:
You're welcome!
I had to improve the verbosity and enhance a bunch of other things.
This time, you'll run just **one** command and share /sdcard/**accd**-S99.log.
`accd -x` hasn't bee implemented until now.
That's the command you'll execute.
[acc_v2021.8.21_(202108210).zip](https://github.com/VR-25/acc/files/7025765/acc_v2021.8.21_.202108210.zip)

nialone, 2021-08-21:
Ready. 
[accd-S99-v21.log](https://github.com/VR-25/acc/files/7026046/accd-S99-v21.log)

VR-25, 2021-08-21:
> Ready. You included all the commands in print? )))

Yes.

> [accd-S99-v21.log](https://github.com/VR-25/acc/files/7026046/accd-S99-v21.log)

Logs suggest that the system has not rebooted.
Try the following (paste and run):
`pkill -9 -f accd; accd -x; sleep 10; acc --set --current -`
Two log files should be generated under /sdcard/.

nialone, 2021-08-21:
> Logs suggest that the system has not rebooted.

I did exactly as instructed. Restart was on the 5 step. 

 
> Try the following (paste and run):
> `pkill -9 -f accd; accd -x; sleep 10; acc --set --current -`

Only one report. [accd-S99-2.log](https://github.com/VR-25/acc/files/7026157/accd-S99-2.log)

VR-25, 2021-08-22:
Your logs don't contain any information related to the _daemon stop_ process.
On my end, I'm getting the expected data in the log file.

1. Run `pkill -9 -f accd; sleep 10; accd -x` and let the daemon run for two minutes or so.
2. While waiting, check whether /scard/accd-S99.log is being updated (it should).
3. Run `acc -x -D stop` (it should trigger a reboot).
4. Share both log files under /sdcard/.

P.S., I expect accd-S99.log to contain the string `exitCode=` towards the end.
This is the missing piece that tells the daemon was indeed stopping just before the reboot.
My log file has that.
Oddly, yours does not.

nialone, 2021-08-22:
Sorry, my report does not contain 'exitcode ='.
ACC calmly looped every '10 Sleep'.
The phone goes to reboot after 'ACC -X -D Stop'.
That is, in the process, without reaching 'exitcode ='.
I see it in every log.

VR-25, 2021-08-22:
That's intriguing!
While accd is running, does `pkill -f accd` also trigger a reboot?
`pkill -f accd` is another way of gracefully stoping accd.

nialone, 2021-08-22:
>  While accd is running, does `pkill -f accd` also trigger a reboot?

Yes.

VR-25, 2021-08-22:
While accd is _not_ running (i.e., after being killed by `pkill -9 -f accd`):
`acc -x -e`
`acc -x -sc -`
`acc -x -sv -`
Each command should generate /sdcard/acc-S99.log.
Share the log only if a reboot is triggered.

nialone, 2021-08-22:
After each 'pkill -9 -f accd'
'acc -x -e' - normal
'acc -x -sc -' - [acc-S99-sc.log](https://github.com/VR-25/acc/files/7028067/acc-S99-sc.log)
 acc -x -sv - normal

nialone, 2021-08-22:
`pkill -9 -f accd` - what is it '-9'?
Maybe replace all `pkill -f accd` on `pkill -9 -f accd` in the code for MediaTek ?
Unfortunately, I did not find PKILL (uninstall.sh), only Kill (accd.sh, release-lock.sh). Is there a difference... )

VR-25, 2021-08-22:
The _cure_ functionality between pkill and kill is essentially the sam: sending signals to processes.
Signal 9 (-9) or SIGKILL forces accd to stop immediately.
That means it's interrupted _ungracefully_, leaving no time for cleanups.

The last log file suggests that flock, kill and pkill commands are not possible culprits.

Since `acc -sc -` triggers a reboot even when accd is not running, using SIGKILL by default for this device will not prevent the reboots.

Last time, a reboot was triggered just after `set +eux`.
Does any of the following trigger it? (Run each as root):
`set +e`
`set +u`
`set +x`
`set +eux`

nialone, 2021-08-22:
There is no reaction. I gained them in a row in the terminal as root.

nialone, 2021-08-22:
> `set +e`
> `set +u`
> `set +x`
> `set +eux`

There is no reaction.

nialone, 2021-08-22:
[image]

VR-25, 2021-08-23:
I'm somewhat convinced that the charging control files available for your device are problematic.
Curiously, they work well on other MTK devices.

Run `pkill -9 -f accd; sed -i /en_/d /data/adb/vr25/acc/charging-switches.txt; rm /data/adb/vr25/acc-data/config.txt`.
Reboot and see if the issue persists.

nialone, 2021-08-23:
After: 
> pkill -9 -f accd; sed -i /en_/d /data/adb/vr25/acc/charging-switches.txt; rm /data/adb/vr25/acc-data/config.txt

Tested:
[image]

Rebooted. [acc-S99.log](https://github.com/VR-25/acc/files/7034726/acc-S99.log)

VR-25, 2021-08-25:
Looking like a memory leak now. Interesting...
It seems as soon as verbose is turned off (set +x), a reboot is triggered.

acca commands runs with verbose off by default.
Does `pkill -9 -f accd; acca -sc -` also trigger a reboot?

VR-25, 2021-09-05:
Revisiting Our _Beloved_ Issue
ACC uses the following charging switches to manage charging status on your system:
`/proc/mtk_battery_cmd/current_cmd "0 0" "0 1" /proc/mtk_battery_cmd/en_power_path 1 0`
Modifying them under certain conditions may be triggering the unwanted reboots.

Kill accd: `pkill -9 -f accd`

Test 1: Set Permissions 10 Times in a Row, First Control File
```
for i in $(seq 10); do
  chmod 0644 /proc/mtk_battery_cmd/current_cmd
done
```
Test 2: Set Permissions 10 Times in a Row, Second Control File
```
for i in $(seq 10); do
  chmod 0644 /proc/mtk_battery_cmd/en_power_path
done
```
Test 3: Write to The First Control Files 10 Times in a Row
```
for i in $(seq 10); do
  echo "0 0" > /proc/mtk_battery_cmd/current_cmd
done
```
Test 4: Write to The Second Control Files 10 Times in a Row
```
for i in $(seq 10); do
  echo 1 > /proc/mtk_battery_cmd/en_power_path
done
```
Test 5: Asynchronously Write to Both Control Files 10 Times in a Row
```
for i in $(seq 10); do
  echo "0 0" > /proc/mtk_battery_cmd/current_cmd & echo 1 > /proc/mtk_battery_cmd/en_power_path
  wait
done
```

nialone, 2021-09-06:
All tests are successful. 
The charging cable is not connected.

VR-25, 2022-02-03:
The upcoming release will auto detect and blacklist unwritable control files and those that trigger unexpected reboots.

closed 2022-02-03 by VR-25