VR-25/acc #108: Add the ability to turn off the notifications "Connect your charger"

masscream, 2021-08-17:
After the last update, my phone started sending me quite annoying notifications "10%-5% to shutdown, please connect your charger." I found it's a new added functionality in case the shutdown capacity is set (have it on 20%). The battery indicator is right enough for me to know, how many percent is remaining and usually when I don't plug my charger in immediately, it could be due to I'm not having one nearby when I'm somewhere out. However these notifications are quite intrusive, displaying every 2 mins or so with a sound and cannot be pushed aside or turned off (only with the phone or complete sound/vibrations shutdown, what completely ruins any "notify me" principle). Is there something I overlooked please in the settings or if I hate these notifications I just have to stop using the shutdown capacity function, which was OK for me the way it was before this update? It would be very helpful to have the ability to control or shutdown these notifications. Thanks

VR-25, 2021-08-17:
The notifications will be disabled by default in the next version.
If I have enough free time until then, they'll also be customizable.

Meanwhile, you can disable those notification by pasting and running the following on terminal:
`su -c sed -i 's/su -lp.*/: \\/' /data/adb/vr25/acc/accd.sh; su -c chmod u+x /data/adb/vr25/acc/accd.sh`

elem089, 2021-08-24:
unfortunately I couldn't copy&paste this, for whatever reason. So I hope I didn't mistype. Response is what looks like a complete shell script, followed by "exit 0"; took like half a minute or so to complete. Is that to be expected, or did I possibly break something? ;)

VR-25, 2021-08-25:
The line patches and restarts accd.
I just tried it again; it worked and did it fast.
The changes are permanent.

P.S., a new release will be up soon.

closed 2021-08-30 by VR-25