VR-25/acc #109: config.txt sometimes becomes empty and breaks ACC

Macusercom, 2021-08-20:
It seems that at some point (somtimes hours, sometimes even days) the config.txt gets emptied (0 Bytes) and ACC failes due to the language string not being available.

It started happening v2021.8.2 or v2021.8.4 and didn't happen on v2021.7.28.1.

I'm using a OnePlus 7T Pro with OOS 11.0.2.1 with op_disable_charge.

Logs are here: https://we.tl/t-eQuNqIvBeJ

Macusercom, 2021-08-20:
[acc-logs-OnePlus7TPro.zip](https://github.com/VR-25/acc/files/7019542/acc-logs-OnePlus7TPro.zip)

VR-25, 2021-08-22:
Investigating...

Taudris, 2021-09-10:
I saw a similar issue during my testing for #112, though it didn't happen on its own, only when I tried to set `run_cmd_on_pause` to a command that happened to have `>` in it, followed by any other config change. (I didn't try a command that didn't have `>` in it.) This emptied config.txt and caused `acc` to stop working.

Macusercom, 2021-09-10:
I only have a command stopping and restarting acc. So far I had the issue once again in the last 2 weeks. I don't run any command with `run_cmd_on_pause`.

VR-25, 2021-09-16:
Try [this](https://github.com/VR-25/acc/files/7176032/acc_v2021.9.16.1_.202109161.zip) build.
`acc -b` undoes the upgrade.

VR-25, 2021-09-16:
[acc_v2021.9.16.2_(202109162).zip](https://github.com/VR-25/acc/files/7176595/acc_v2021.9.16.2_.202109162.zip)

VR-25, 2021-09-16:
Refer to issue #106 for newer builds.

closed 2021-09-19 by VR-25