VR-25/acc #111: Two acc daemons are running

axonasif, 2021-08-30:
Hi, I noticed that there are two instances of acc daemon running in my device. Is this normal?
[image]

Here is the tarballed logs dir:
[logs.tar.gz](https://github.com/VR-25/acc/files/7077708/logs.tar.gz)

Additional info:
At first I had only installed aac from magisk and later installed accA. But then I removed both and installed aacA only which installed the magisk module automatically. In all these senarios I noticed the same behavior of two daemons to be running.

Thanks.

VR-25, 2021-08-30:
The latest acc version has this fixed.

axonasif, 2021-08-30:
> The latest acc version has this fixed.

It's still happening, the second daemon somehow spawns after a few seconds of boot. Now I only use cli acc, uninstalled accA and freshly reinstalled acc module from magisk.

VR-25, 2021-08-30:
Right after spotting the issue again, please share the output of `pgrep -fl accd` and the log archive generated by `acc -le`.
Both commands must run as root.

axonasif, 2021-08-30:
> Right after spotting the issue again, please share the output of `pgrep -fl accd` and the log archive generated by `acc -le`.
> Both commands must run as root.

pgrep output:
```
root@localhost /d/d/c/f/home# pgrep -fl accd
188 accdet
190 accdet_eint
1113 accd.sh
6817 accd.sh
```

Acc dumped logs:
[logs.tar.gz](https://github.com/VR-25/acc/files/7078443/logs.tar.gz)

> I have a suggestion, it would be better if you guys would use standard gzip for tar compression instead of bz. GitHub too doesn't allow uploading that file format.

closed 2021-09-19 by VR-25