VR-25/acc #112: Possible battery idle solution for OnePlus devices

Taudris, 2021-08-31:
I have a OnePlus 6. I used FX File Explorer to poke at the different files by hand while ACC was disabled, and landed on a sequence that I suspect works consistently. The charging switch string would be this if ACC supported more than 2 charging switches:

    battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1

Then the OEM workaround should not be needed.

`battery/op_disable_charge` seems to enter battery idle mode correctly when:
* It is changed while a charger is not plugged in, then a charger is plugged in after.
* It is changed while a charger is plugged in and `battery/charging_enabled` is 0.

When attempting to change `battery/op_disable_charge` while the phone is plugged in and `battery/charging_enabled` is 1, it does stop charging, but almost immediately resumes charging, first regular, then fast.

These observations were made using the OnePlus charger that my phone came with.

VR-25, 2021-08-31:
Perhaps we don't need a third charging switch.
Try the following:

`accd.; acc --set run_cmd_on_pause="echo 1 \\> battery/charging_enabled"; acc --set s="battery/charging_enabled 0 0 battery/op_disable_charge 0 1"`

Test idle mode manually with `acc -d`.

MrZell, 2021-09-01:
Maybe this helps, maybe it doesn't, but battery idle mode works for me on my OnePlus 8T.

I have to use the "battery/op_disable_charge 0 1 battery/input_suspend 0 0" charging switch and a USB-PD charger, not the 65W OEM one. If I use the OEM charger, the current limit, voltage limit, battery % limit, and idle mode all don't work using that switch. If I swap over to use the "battery/charging_enabled 1 0" switch, the battery % limit does work as it should (even on the 65W OEM charger), but none of the other features mentioned above do.

Every OnePlus device I have used has always been quite particular when it comes to charging so it wouldn't surprise me if there isn't a switch that would just work for all chargers.

Taudris, 2021-09-03:
I haven't been able to get `acc` commands working correctly via Termux or Terminal Emulator (both from F-Droid). I'm sometimes able to get to the `acc` menu. I usually get weird errors about `grep` or `pgrep` `Bad address`, or `acc` ignores the actual command and goes and creates a log file archive instead. I'm using the Magisk version of `acc`. I tried reinstalling it from Magisk, installing `v2021.8.31` from `AccA` (itself installed from F-Droid), rebooting in between, still no dice.

I was able to remove the OnePlus stuff from `oem-custom.sh` and edit `misc-functions.sh` in a very hacky way to get it to set the settings I need for entering battery idle. (See screenshots.) It works with both the OnePlus Dash charger and with a USB-C port on a dock. I'm able to plug it in, wait for it to go to battery idle, unplug it, plug it in again, and it begins charging, then re-enters battery idle. But now the `acc` daemon is regularly killing itself..? I don't know what I did wrong. I'll mess with it some more; I need something to tide me over.

I also just noticed that the second charging switch only works when disabling charging. It's ignored on the enabling side. @VR-25, are you sure you don't want to write a function to iterate over all the charging switches? Or maybe a better solution would be to provide a custom command to pause and unpause charging.

Termux:
[image]

Terminal Emulator:
[image]

Modified script:
[image]

Success! (Until `accd` dies, anyway. I don't know what triggers it and the log in AccA doesn't show a reason.)
[image]

Taudris, 2021-09-03:
Running `acc --set run_cmd_on_pause="echo 1 \\> battery/charging_enabled"` followed by `acc --set s="battery/charging_enabled 0 0 battery/op_disable_charge 0 1"` causes `config.txt` to become empty, which causes `acc` to stop working and only spit out log files. Deleting `config.txt` allows `acc --reset` to work. Running those commands in the opposite order prevents this. I don't know if it matters whether `charging_switch` is set before running `acc --set run_cmd_on_pause`; I've had this issue occur even after selecting a charging switch in AccA.

Anyway, since I figured that out, `acc -d` does indeed trigger battery idle, and as expected, `battery/charging_enabled=1` and `battery/op_disable_charge=1`.

`acc -e` does not re-enable charging, nor does unplugging and replugging the charger. In this state, `battery/charging_enabled=0` and `battery/op_disable_charge=0`. This makes sense, though, given there's no `run_cmd_on_resume`.

VR-25, 2021-09-03:
@Taudris, I will modify the logic so that an _unlimited_ number of charging switches is supported.
For the command errors, try the build below.
If it does not help, reproduce some of the issues again and share a log archive right after (`acc -le`).
[acc_v2021.9.1_(202109010).zip](https://github.com/VR-25/acc/files/7105629/acc_v2021.9.1_.202109010.zip)

Taudris, 2021-09-03:
Still breaks.

[acc-logs-OnePlus6.tar.gz](https://github.com/VR-25/acc/files/7107121/acc-logs-OnePlus6.tar.gz)

[image]

VR-25, 2021-09-03:
> Still breaks.
> 
> [acc-logs-OnePlus6.tar.gz](https://github.com/VR-25/acc/files/7107121/acc-logs-OnePlus6.tar.gz)

This is happening because you config is null!
Run `rm /data/adb/vr25/acc-data/config.txt` to reset it.

Here's a new build.
It supports an unlimited number of charging switches and has your group of 3 built-in.
See how things go.
[acc_v2021.9.3.1_(202109031).zip](https://github.com/VR-25/acc/files/7108384/acc_v2021.9.3.1_.202109031.zip)

Taudris, 2021-09-04:
Thanks for the new build! I'll try it out when I have a bit more brain juice to spare.

I figured out the null config problem and fix already, and tried to document it above, but my excess verbosity clearly got in the way... Sorry about that.

You can cause `acc` to write out a null config like so:
1. Run `acc --set run_cmd_on_pause="echo 1 \\> battery/charging_enabled"`. config.txt now contains the setting, with some kind of escaping. (I don't know nearly enough shell script/Linux CLI to say if it's correct. I don't have an example handy right now.)
2. Change the charging switch, or very likely, make any other config change. config.txt is now blank.

Taudris, 2021-09-06:
Alright, I poked at this some more using v2021.9.5. Using the new charge switch didn't work at all until I added `sleep 1` in the loop of `flip_sw`. It would work with smaller values, but it would have to go through `disable_charge` a second time (or so I assume, since it takes 5+ seconds) to actually go to battery idle. 1 second seems to reliably go to battery idle without taking 5+ seconds.

I'm not able to see the new charge switch anywhere but in the charge switch text file. It doesn't show in AccA and I don't know how to get `acc` to print a list or allow me to select one. Setting the charge switch myself worked fine.

VR-25, 2021-09-07:
So, we have to wait a bit between writes.
Toggle switch 1, wait, toggle switch 2, wait... and so on.
Did I get that right?

To simplify testing across builds, you can override `flip_sw()` and any other function (and even some global variables).
Create `/data/adb/vr25/acc-data/plugins/filename.sh` and place your custom functions there.

Doesn't `acc -ss:` print a list?
`acc -ss` is the selection wizard.

Taudris, 2021-09-07:
Yes, that's correct, wait between writes.

I thought I had a hackishly edited version working without waits at one point, but I haven't been able to recreate it since. It just tries and fails to enter battery idle over and over (charging gets disabled and re-enabled, vibrating the device each time). The only thing that seems to work is adding the delay. 😭

Excellent tip for replacing functions! I'll try that out and see how it goes.

I didn't know about `acc -ss:`, I normally just use AccA. I just tried it. It only prints 9 switches, and it doesn't include the new OnePlus switch. The AccA list is the exact same. If I omit the `:` to use the menu, there is one additional switch in the menu, `3700`.

I wonder if my install is somehow using a mix of old and new `acc` files..? Any thoughts on how to check?

VR-25, 2021-09-07:
> I wonder if my install is somehow using a mix of old and new `acc` files..? Any thoughts on how to check?

This is unlikely...
You can inspect the contents of `/data/adb/modules/acc/`.
Have you rebooted after the upgrade? (Just to see if it makes a difference.)
Also, see if `/de/.vr25/acc/ch-switches` contains the triple switch.
`3700` is meant for controlling charging using voltage limits over charging switches (_3700_ mV to stop, _default_ to resume).

Taudris, 2021-09-07:
Just rebooted. `/dev/.vr25/acc/ch-switches` does not contain the triple switch. It has a modified date from during the reboot, and it only has these 5 lines:

* `battery/charging_enabled 1 0`
* `battery/input_suspend 0 1`
* `battery/op_disable_charge 0 1`
* `battery/op_disable_charge 0 1 battery/input_suspend 0 0`
* `dc/input_suspend 0 1`

VR-25, 2021-09-07:
As I suspected...
Will work on this and on the automation of `sleep 1`.

VR-25, 2021-09-07:
[acc_v2021.9.7_(202109070).zip](https://github.com/VR-25/acc/files/7122149/acc_v2021.9.7_.202109070.zip)
- Any path in `acc/charging-switches.txt` can contain wildcards.
- General fixes & optimizations
- When a charging switch group is in use, wait 1 second between writes.

Undo all modifications/workarounds before upgrading.
Test idle mode, `acc -ss:` and `acc -ss`.

Taudris, 2021-09-09:
`acc -ss:` now shows a bunch of charging switches on one line. This is reflected in `/dev/.vr25/acc/ch-switches`.

Colo-Thor, 2021-09-09:
OnePlus8T acc -- info STATUS=Not charging is idle mode？Charge counter decreasing gradually

VR-25, 2021-09-09:
> `acc -ss:` now shows a bunch of charging switches on one line. This is reflected in `/dev/.vr25/acc/ch-switches`.

I was able to reproduce and fix that.
[acc_v2021.9.9_(202109090).zip](https://github.com/VR-25/acc/files/7137366/acc_v2021.9.9_.202109090.zip)

Taudris, 2021-09-09:
* Idle is working.
* Resume from idle is working.
* `ch-switches` is no longer all on one line.
* `acc -ss:`, `acc -ss`, and `ch-switches` still show the old list.

Contents of `ch-switches` (note that the first line is blank):

```

battery/charging_enabled 1 0
battery/input_suspend 0 1
battery/op_disable_charge 0 1
battery/op_disable_charge 0 1 battery/input_suspend 0 0
dc/input_suspend 0 1
```

Colo-Thor, 2021-09-09:
> > OnePlus8T acc -- info STATUS=Not charging is idle mode？Charge counter decreasing gradually
> 
> Yes, this is idle mode.
> `STATUS=Not charging` is the main indicator of it.

in idle mode,will the battery capacity still decrease?

VR-25, 2021-09-09:
@Taudris, could you share `/data/adb/vr25/acc-data/logs/init.log`?

VR-25, 2021-09-09:
> in idle mode,will the battery capacity still decrease?

Yes, just like when you store a battery.
It takes a long time, obviously.

I'll add the following to the documentation:

Idle Mode and Alternatives

1. Charging switch that supports idle mode (the obvious winner).
Note that self discharge is a thing.
This is as if the battery were physically disconnected.
Extremely slow discharge rate is expected.

2. charging_switch=0
If current fluctuates, also set current_workaround=true (only takes affect after a reboot).
If this method works, the behavior is exactly the same as 1.

3. charging_switch=3920: only works on devices that actually support voltage control.
Unlike regular idle mode, this maintains 3920 mV (the sweet spot) indefinitely.
This is not good with higher voltages.
We're trying to minimize battery stress as much as possible.
Maintaining a voltage higher than 3920 for a long time is not recommended.

4. acc 3920
This is short for acc 3920 3870 (a 50 mV difference).
It tries to maintain 3920 mV without voltage control support.
Yes, it's definitely not a joke.
This works with regular charging switches and voltage readings.

5. acc 45 44
I prefer this over acc 50 49, since voltage and capacity (%) do not have a linear relationship.
Voltage varies with temperature, battery chemistry and age.
Considering those variables, acc 45 44 is more optimal than acc 50 49.
50% is more likely to translate to voltage values higher than 3920 mV.

Colo-Thor, 2021-09-09:
> Yes, just like when you store a battery.
> It takes a long time, obviously.
>
> I'll add the following to the documentation:
>
> Idle Mode and Alternatives
>
>     1. Charging switch that supports idle mode [...]

Thanks for the answer, my phone is like disconnecting the charger in this mode, and it may need to customize the configuration

Taudris, 2021-09-09:
@Colo-Thor That's exactly what @VR-25 and I are working on fixing 😃

@VR-25 Log file as requested: [init.log](https://github.com/VR-25/acc/files/7137827/init.log)

Colo-Thor, 2021-09-09:
> @Colo-Thor That's exactly what @VR-25 and I are working on fixing 😃
> 
> @VR-25 Log file as requested: [init.log](https://github.com/VR-25/acc/files/7137827/init.log)

Okay, I didn’t look carefully, sorry

Taudris, 2021-09-09:
@Colo-Thor If you install `acc_v2021.9.9_(202109090).zip`, you can set a custom charging switch to get idle mode working right now, like so:

```
acc --set s="battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1"
```

VR-25, 2021-09-09:
> * `acc -ss:`, `acc -ss`, and `ch-switches` still show the old list.
>
> Contents of `ch-switches` (note that the first line is blank):
>
> ```
>
> battery/charging_enabled 1 0
> battery/input_suspend 0 1 [...]

[acc_v2021.9.9.1_(202109091).zip](https://github.com/VR-25/acc/files/7138273/acc_v2021.9.9.1_.202109091.zip)

Colo-Thor, 2021-09-10:
> @Colo-Thor如果你安装了`acc_v2021.9.9_(202109090).zip`，你可以设置一个自定义充电开关来让空闲模式立即工作，如下所示：
> 
> ```
> acc --set s="battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1"
> ```

does work
> ```
> OnePlus8T:/ $ dumpsys battery
> Current Battery Service state:
> AC powered: true
> USB powered: false
> Wireless powered: false
> Max charging current: 1000000
> Max charging voltage: 9000000
> Charge counter: 3364752
> status: 4
> health: 2
> present: true
> level: 99
> scale: 100
> voltage: 4343
> temperature: 336
> technology: Li-ion
> fastChargeStatus: false
> ```

Taudris, 2021-09-10:
With `acc_v2021.9.9.1_(202109091).zip`, `ch-switches` is now:
```

battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1
battery/charging_enabled 1 0
battery/input_suspend 0 1
battery/op_disable_charge 0 1
battery/op_disable_charge 0 1 battery/input_suspend 0 0
dc/input_suspend 0 1
```

AccA now shows the new charging switch without me having to add it manually. I think that's a wrap!

@Colo-Thor It looks like your battery charged to 100%. This does not demonstrate battery idle mode.

Colo-Thor, 2021-09-10:
> With `acc_v2021.9.9.1_(202109091).zip`, `ch-switches` is now:
>
> ```
>
> battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1
> battery/charging_enabled 1 0 [...]

battery charged to 95%：
```
OnePlus8T:/ # dumpsys battery
Current Battery Service state:
  AC powered: true
  USB powered: false
  Wireless powered: false
  Max charging current: 3000000
  Max charging voltage: 9000000
  Charge counter: 3323441
  status: 4
  health: 2
  present: true
  level: 95
  scale: 100
  voltage: 4310
  temperature: 307
  technology: Li-ion
  fastChargeStatus: false
```
current problem now is that accd will stop running：
```
Advanced Charging Controller v2021.9.9 (202109090)
Copyright 2017-2021, VR25
GPLv3+

(i) accd is not running
```

VR-25, 2021-09-10:
> current problem now is that accd will stop running：
> ```
> Advanced Charging Controller v2021.9.9 (202109090)
> Copyright 2017-2021, VR25
> GPLv3+
> 
> (i) accd is not running
> ```

Share the output of `acc -l tail`.
It may reveal why accd is stopping.
My guess is that all charging switches are failing (exit code 7).

Colo-Thor, 2021-09-10:
log file is /dev/.vr25/acc/[acc-OnePlus8T.log](https://github.com/VR-25/acc/files/7141966/acc-OnePlus8T.log)?
now accd is not running, prior to this, the charger was connected, but frequently charging and powering off several times

Colo-Thor, 2021-09-10:
delete old log, accd stop runnning repeat
[acc-OnePlus8T.log](https://github.com/VR-25/acc/files/7142019/acc-OnePlus8T.log)

VR-25, 2021-09-10:
@Colo-Thor, `accd` is stopping because all charging switches are failing.
What's the output of `acc -t`?

Colo-Thor, 2021-09-10:
> @Colo-Thor, `accd` is stopping because all charging switches are failing.
> What's the output of `acc -t`?
```
OnePlus8T:/ # acc -t

(!) Charger must be plugged to continue...
(i) Alright, this may take a minute or so...

(i) [battery/charging_enabled 1 0] works
- battIdleMode=true

(i) [battery/input_suspend 0 1] works
- battIdleMode=true

(i) [battery/op_disable_charge 0 1] works
- battIdleMode=true

(i) [battery/op_disable_charge 0 1 battery/input_suspend 0 0] works
- battIdleMode=true

(!) [dc/input_suspend 0 1] won't work

(i) [op_charger/charge_enabled 1 0] works
- battIdleMode=true
```

VR-25, 2021-09-10:
This could mean those charging switches have a delayed response.
Run `acc -ss`, pick one of the working switches and you should be good.
`accd` will no longer stop due to switch failure.

Colo-Thor, 2021-09-10:
seems ok
```
OnePlus8T:/ # acc -c cat

configVerCode=202108120
capacity=(0 60 70 75 false)
temperature=(40 60 90 65)
cooldownRatio=( )
cooldownCurrent=
cooldownCustom=()
resetBattStats=(false false)
chargingSwitch=(0)
applyOnBoot=()
applyOnPlug=()
maxChargingCurrent=()
maxChargingVoltage=()
language=en
runCmdOnPause=()
ampFactor=
voltFactor=
loopCmd=()
prioritizeBattIdleMode=false
currentWorkaround=false

OnePlus8T:/ # acc --info | grep STATUS
  CHG_PROTECT_STATUS=0
  FASTCHG_STATUS=0
  FASTCHG_STATUS_IS_OK=1
  FASTCHG_TEMP_STATUS=0
  STATUS=Not charging

OnePlus8T:/ # dumpsys battery
Current Battery Service state:
  AC powered: false
  USB powered: false
  Wireless powered: false
  Max charging current: 0
  Max charging voltage: 0
  Charge counter: 3124799
  status: 4
  health: 2
  present: true
  level: 89
  scale: 100
  voltage: 4218
  temperature: 324
  technology: Li-ion
  fastChargeStatus: false
```

Taudris, 2021-09-10:
@Colo-Thor It looks like your phone isn't in battery idle mode. Is the battery % going down from the limit you set? When my OnePlus 6 is in battery idle, `dumpsys battery` shows `AC powered: true`. I expect that yours should show the same.

In `acc -t`, the first switch should be:

```
battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1
```

You need to install `acc_v2021.9.9.1_(202109091).zip` for that switch to show up automatically:

> [acc_v2021.9.9.1_(202109091).zip](https://github.com/VR-25/acc/files/7138273/acc_v2021.9.9.1_.202109091.zip)

Colo-Thor, 2021-09-10:
> @Colo-Thor您的手机似乎未处于电池空闲模式。电池百分比是否低于您设置的限制？当我的 OnePlus 6 处于电池空闲状态时，`dumpsys battery`显示`AC powered: true`。我希望你的应该显示相同。
> 
> 在 中`acc -t`，第一个开关应该是：
> 
> ```
> battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1
> ```
> 
> 您需要安装`acc_v2021.9.9.1_(202109091).zip`该开关才能自动显示：
> 
> > [acc_v2021.9.9.1_(202109091).zip](https://github.com/VR-25/acc/files/7138273/acc_v2021.9.9.1_.202109091.zip)

I will observe whether the battery capacity will decrease
```
OnePlus8T:/ # accd,

(i) accd v2021.9.9.1 (202109091) is running (PID 10429)

OnePlus8T:/ # acc -c cat | head -n 20

configVerCode=202108120
capacity=(0 60 70 75 false)
temperature=(40 60 90 65)
cooldownRatio=( )
cooldownCurrent=
cooldownCustom=()
resetBattStats=(false false)
chargingSwitch=(battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1 --)
applyOnBoot=()
applyOnPlug=()
maxChargingCurrent=()
maxChargingVoltage=()
language=en
runCmdOnPause=()
ampFactor=
voltFactor=
loopCmd=()
prioritizeBattIdleMode=false
currentWorkaround=false

OnePlus8T:/ # acc --info | grep STATUS
  CHG_PROTECT_STATUS=0
  FASTCHG_STATUS=0
  FASTCHG_STATUS_IS_OK=1
  FASTCHG_TEMP_STATUS=0
  STATUS=Not charging

OnePlus8T:/ # dumpsys battery
Current Battery Service state:
  AC powered: true
  USB powered: false
  Wireless powered: false
  Max charging current: 3000000
  Max charging voltage: 9000000
  Charge counter: 2956838
  status: 4
  health: 2
  present: true
  level: 83
  scale: 100
  voltage: 4173
  temperature: 354
  technology: Li-ion
  fastChargeStatus: false
```

Colo-Thor, 2021-09-10:
> I will observe whether the battery capacity will decrease
>
> ```
> OnePlus8T:/ # accd,
>
> (i) accd v2021.9.9.1 (202109091) is running (PID 10429)
>
> OnePlus8T:/ # acc -c cat | head -n 20 [...]

It should be fine. My phone is always playing online videos, and the battery capacity is always at 83%.

Taudris, 2021-09-10:
Excellent!

@VR-25:

* It seems that `acc_v2021.9.9.1_(202109091).zip` is working perfectly for both myself and @Colo-Thor.
* One of the things you can try having `acc` check (if it isn't already) for determining whether a device is in battery idle is `dumpsys battery` should say `AC powered: true`.

VR-25, 2021-09-10:
Here's one more:

[**v2021.9.10 (202109100)**](https://github.com/VR-25/acc/files/7143322/acc_v2021.9.10_.202109100.zip)
- Additional charging switches - the database is more concise with the extensive use of wildcards.
- `current_workaround` no longer requires a reboot (just `accd --init`).
- General fixes & optimizations
- Updated documentation (mainly tips > idle mode and alternatives)
- Upgrade rollback feature (`-b|--rollback` or wizard option `f`)

This will be released today if there are no major issues.
So far so good on my end.

Colo-Thor, 2021-09-10:
> Here's one more:
>
> [**v2021.9.10 (202109100)**](https://github.com/VR-25/acc/files/7143322/acc_v2021.9.10_.202109100.zip)
>
>     * Additional charging switches - the database is more concise with the [...]

Okay, try it now and give feedback if have any questions

Taudris, 2021-09-10:
It's working for me.

Colo-Thor, 2021-09-10:
I tried the 65w charger that comes with the phone, it will charge and power off frequently, I used a normal slow charger before
~~~
:/ # dumpsys battery
Current Battery Service state:
  AC powered: false
  USB powered: false
  Wireless powered: false
  Max charging current: 0
  Max charging voltage: 0
  Charge counter: 2868119
  status: 4
  health: 2
  present: true
  level: 75
  scale: 100
  voltage: 4086
  temperature: 341
  technology: Li-ion
  fastChargeStatus: false
:/ # dumpsys battery
Current Battery Service state:
  AC powered: true
  USB powered: false
  Wireless powered: false
  Max charging current: 3000000
  Max charging voltage: 5000000
  Charge counter: 2868114
  status: 2
  health: 2
  present: true
  level: 75
  scale: 100
  voltage: 4109
  temperature: 341
  technology: Li-ion
  fastChargeStatus: false
~~~

Taudris, 2021-09-11:
That's disappointing. That's exactly what I was hoping all this work would fix, and it does fix it for me. You're sure you're using the new long charging switch? Maybe automatic mode isn't picking the right one, so try selecting it manually.

```
battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1
```

VR-25, 2021-09-12:
[acc_v2021.9.12_(202109120).zip](https://github.com/VR-25/acc/files/7149752/acc_v2021.9.12_.202109120.zip)
@Taudris et al., I made enhancements to this one.
See how it behaves.

Taudris, 2021-09-12:
This version enters and exits idle mode significantly faster, I assume chiefly thanks to the new `write` function.

@Colo-Thor, give this one a shot with your 65W charger.

Colo-Thor, 2021-09-13:
The slow charger works normally, the 65w charger will charge slowly if it exceeds the limit
```
OnePlus8T:/ # dumpsys battery
Current Battery Service state:
  AC powered: false
  USB powered: false
  Wireless powered: false
  Max charging current: 0
  Max charging voltage: 0
  Charge counter: 2458895
  status: 4
  health: 2
  present: true
  level: 80
  scale: 100
  voltage: 4139
  temperature: 318
  technology: Li-ion
  fastChargeStatus: false
OnePlus8T:/ # acc --info |grep NOW
  CHARGE_NOW=0
  CHARGE_NOW=5025
  CHARGE_NOW_RAW=2296504
  CURRENT_NOW=71000
  POWER_NOW=38792184
  TIME_TO_FULL_NOW=7140
  VOLTAGE_NOW=4135000
  CURRENT_NOW=0.071 Amps
  VOLTAGE_NOW=4.135 Volts
  POWER_NOW=0.293585 Watts

OnePlus8T:/ # dumpsys battery
Current Battery Service state:
  AC powered: false
  USB powered: false
  Wireless powered: false
  Max charging current: 0
  Max charging voltage: 0
  Charge counter: 2457386
  status: 2
  health: 2
  present: true
  level: 81
  scale: 100
  voltage: 4160
  temperature: 316
  technology: Li-ion
  fastChargeStatus: false
OnePlus8T:/ # acc --info |grep NOW
  CHARGE_NOW=2458173
  CHARGE_NOW=4950
  CHARGE_NOW_RAW=2296192
  CURRENT_NOW=-170000
  POWER_NOW=38799490
  TIME_TO_FULL_NOW=7620
  VOLTAGE_NOW=4167000
  CURRENT_NOW=0.17 Amps
  VOLTAGE_NOW=4.167 Volts
  POWER_NOW=0.70839 Watts

VR-25, 2021-09-13:
> The slow charger works normally, the 65w charger will charge slowly if it exceeds the limit

This is a kernel issue.
OnePlus' wicked fast charging is known for overriding limits.
There may be an undiscovered combination of switches that is able to stop this.

@Taudris, could you check whether the switches still work after replacing `.33` with `.1` in `misc-functions.sh:373`?
`sed -i s/33/1/ /data/adb/vr25/acc/misc-functions.sh`

And perhaps removing the `3` from line `371`, so that the `for` loop only has 2 iterations.

VR-25, 2021-09-13:
[Here's](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip) a build that implements both changes and additional fixes.

Taudris, 2021-09-13:
Still working well for me. I entered and exited idle several times just to make sure it worked each time, and I didn't have any issues.

MrZell, 2021-09-13:
I'll throw this build on my 8T this evening and see what happens with the 65W Warp Charger, both with a 3A non-emarked cable and the OEM cable. 

My current experience is that the OEM charger always kicks the phone into Warp Charge and ignores the limits in AccA as it sees fit. On 2 or 3 occasions I have seen the idle mode work though and stop charging at the 79-80% mark as intended. Using a 3A cable limits charging to 27W (9V @3A) but often drops to 18W (9V @2A) which is where I have my current limit set to. I think it is just a coincidence though as using a 5A emarked cable brings the charging up to 24W (9V @2.5A) in the same scenario. We'll see if the v2021.9.13 build changes that!

ezbik, 2021-09-14:
Hello,

- accd v2021.9.13 (202109130)
- **OnePlus 6T A6013**

I tried:

```
accd.
acc --set s="battery/charging_enabled 0 0 battery/op_disable_charge 0 1 battery/charging_enabled 1 1"
```

```
# acc -ss

(i) Known charging switches
1) Automatic
2) battery/charging_enabled 1 0
3) battery/input_suspend 0 1
4) battery/step_charging_enabled 1 0
5) dc/input_suspend 0 1
6) 0
7) 250
8) 350
9) 500
10) 3700
11) Exit
```

```
# acc -t

(!) Charger must be plugged to continue...
(i) Alright, this may take a minute or so...
(!) [battery/charging_enabled 1 0] won't work
(!) [battery/input_suspend 0 1] won't work
(!) [battery/step_charging_enabled 1 0] won't work
(!) [dc/input_suspend 0 1] won't work
```

`acc -i` => https://paste.ee/p/xQGbq

VR-25, 2021-09-15:
[acc_v2021.9.15.1_(202109151).zip](https://github.com/VR-25/acc/files/7172169/acc_v2021.9.15.1_.202109151.zip)

VR-25, 2021-09-16:
https://github.com/VR-25/acc/files/7176032/acc_v2021.9.16.1_.202109161.zip

VR-25, 2021-09-16:
[acc_v2021.9.16.2_(202109162).zip](https://github.com/VR-25/acc/files/7176595/acc_v2021.9.16.2_.202109162.zip)

VR-25, 2021-09-16:
Refer to issue #106 for newer builds.

closed 2021-09-19 by VR-25