VR-25/acc #113: acc buggy on redmi note 6 pro(tulip)

kevin01523, 2021-08-31:
acc seems too buggy for my device on the latest update
acc -wizard is bugged
acc doesnt show current
acc seems to control current fine via checking on ru.andr7e.deviceinfohw.pro app on its battery tab

VR-25, 2021-08-31:
What's wrong with the wizard?

ACC is not showing the correct current value because your device is reporting two - yes, two `current_now` values - negative and positive.
This breaks the logic.
I'll have to rewrite part of it.

The AccA issue is a bit weird.
It does not seem to be related to acc.
Otherwise, the app would not be able to read the config at all.
Below "Custom Settings", the config is correct - meaning, acc is ok.
This is under investigation.
I haven't tested the previous AccA version with the latest acc.

kevin01523, 2021-08-31:
oh the config seems to have disappeared from /data/media/0/documents folder i think that causes the config issue cannot be read by acca
it now defaults into /data/adb/vr25 i didn't reset the config before upgrade but it seemed to have been moved there

VR-25, 2021-08-31:
The config is not the issue.
The path changed a while ago for security reasons.

Another user just reported that the latest acc is working with the previous AccA.
The only issue is "CURRENT_NOW=0 mA".

`acc -wizard` is an invalid command.
What you want is just `acc` (without the `-wizard` part).

kevin01523, 2021-08-31:
oh i guess acca is having issues reading the config from its default location now it was ok on its previous version oh well tnx for the info

kevin01523, 2021-08-31:
thanks it works now ill test for more bugs