VR-25/acc #114: doesn't stop charging

kevin01523, 2021-08-31:
i set a limit where it should stop charging on 70% but still continue to charge. although it doesn't appear to be charging at a glance but on closer look when i plug the charger it changes the current but doesn't notify

VR-25, 2021-09-01:
Share the output of `acc -t` (latest acc).

kevin01523, 2021-09-01:
`.../files/home # acc -t

(!) Charger must be plugged to continue...
(i) Alright, this may take a minute or so...

(i) [battery/charging_enabled 1 0] works
- battIdleMode=false

(i) [battery/input_suspend 0 1] works
- battIdleMode=false

(!) [dc/input_suspend 0 1] won't work`

VR-25, 2021-09-01:
Run `acc -s s="battery/input_suspend 0 1 --"` and see whether charging actually stops.

kevin01523, 2021-09-01:
it only stops by a second then continue to to charge only removed or updates the battery and charger light indicator

before
[image]

after

[image]

VR-25, 2021-09-01:
Run `acc -s s=` to revert to _automatic_ charging switch.
Since the charging status actually changes, this seems to be a kernel issue.
ACC actually _thinks_ charging stopped - because _battery/status_ says so.
The kernel is _lying_.

Try current and voltage based charging control.
Current: `acc -s s=0`
Voltage: `acc -s s=3700`
Test one at a time.

VR-25, 2021-09-01:
Are you sure the previous acc version actually works?

kevin01523, 2021-09-01:
yes its on automatic on my end as always
then it defaults into battery charging enabled switch with cycling allowed

i think i suspect the issue is on the new version
once the cycling through switches is enabled after testing the switches somehow
battery/input_suspend  gets used /prioritized even if
the default one is battery/charging_enabled 1 0 ill test more i guess
on previous version it may skip a percent or two(rarely) but eventually the charging stops but now it can charge into 100 unknowingly

kevin01523, 2021-09-03:
> Are you sure the previous acc version actually works?

Yep tried and tested same setup update 21 8 4 properly stops charging at auto
Or won't charge after replugged the charger

8 31
Somehow won't stop charging(after a few more test both switches seems affected charging enabled and suspend) after passes the limit and will charge up to 100 with the charging status as false but you can see it charging based on current and what kind of charge if its taper or fast charging

programminghoch10, 2021-09-03:
Hey, I have the same issue on willow, old versions worked flawlessly, new one does not stop charging. I first thought it might be a problem with ACCA, so I'll test to downgrade ACCA only first.

VR-25, 2021-09-03:
Try [this](https://github.com/VR-25/acc/files/7105629/acc_v2021.9.1_.202109010.zip) new build.

programminghoch10, 2021-09-03:
Short test seems to work, I'll try in my usual cycle this night.

VR-25, 2021-09-03:
[Here's](https://github.com/VR-25/acc/files/7108384/acc_v2021.9.3.1_.202109031.zip) a newer build.

programminghoch10, 2021-09-03:
@VR-25 the ACC Daemon crashed on me again. I'll send the log on Sunday.

kevin01523, 2021-09-04:
Did a quick test it does stop after hitting the stop charging at value but once I tried to toggle off the automatic cycling through switches and save the phone seems to start charging without changingvits status from discharging.

I used the last build provided

programminghoch10, 2021-09-05:
[acc-logs-willow.tar.gz](https://github.com/VR-25/acc/files/7112142/acc-logs-willow.tar.gz)
Acc again crashed just while charging in the low 30%, it didn't even nearly hit the stop limit.

VR-25, 2021-09-05:
A [new](https://github.com/VR-25/acc/releases/tag/v2021.9.5) release is up.

kevin01523, 2021-09-07:
here are some new logs i have encountered a few weird bugs
like config getting reset to previous setting or config changes doesn't save. on acca had to re edit and save a few times.it doesn't always happen but it happens.

it tries to charge for a second then stops(might be the result of different config im seeing and the actual config being used via acca)

i guess you already know that cooldown is broken as it stops and resume in less than 5-10 seconds intervals contrary to the config

programminghoch10, 2021-09-09:
For me everything works now. It ran several cycles without any problems.

kevin01523, 2021-09-10:
I think the only really broken one is the cooldown time
The config daving i think its a acca bug where there's two simultaneous commands setting into the config and it hangs

VR-25, 2021-09-10:
I still could not reproduce the cooldown issue.
Cooldown is working well for me.

Here's a new prerelease:
https://github.com/VR-25/acc/files/7138273/acc_v2021.9.9.1_.202109091.zip

kevin01523, 2021-09-10:
Here a log from just before the prerelease gonna upload the  prerelease log after

From what i observed cooldown goes haywire if i use a fast charger with qc 3.0 with a current limited by acc on my other charger it goes fine kinda

kevin01523, 2021-09-10:
It crashes once i plug in a charger any charger will do tested 3 of em

VR-25, 2021-09-10:
It seems AccA is causing problems.
I've not been using it lately.
See if acc alone has any major issue.

VR-25, 2021-09-10:
[acc_v2021.9.10_(202109100).zip](https://github.com/VR-25/acc/files/7143322/acc_v2021.9.10_.202109100.zip)

kevin01523, 2021-09-10:
Crashes when a charger is plugged i did set things via command so no acca
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7143606/acc-logs-tulip.tar.gz)

VR-25, 2021-09-10:
[acc_v2021.9.10.1_(202109101).zip](https://github.com/VR-25/acc/files/7143873/acc_v2021.9.10.1_.202109101.zip)

kevin01523, 2021-09-10:
Crashes when a charger is plugged i did set things via command so no acca
[Uploading acc-logs-tulip.tar.gz…]()

kevin01523, 2021-09-10:
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7144180/acc-logs-tulip.tar.gz)
Cooldown seems to work now i haven't tried or set any new value after updating acc i guess ill observe and charge it for a few cycle the. Upload logs

kevin01523, 2021-09-11:
It didn't report that its charging but it is suddenly
After changing the max current not sure what really happened
Restarting acc from term fixed it and the switches git set to auto
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7147152/acc-logs-tulip.tar.gz)
Logs for details

VR-25, 2021-09-11:
I noticed that you're using `cooldownCurrent=700`.
Perhaps your device does not support this. Hence, cooldown is failing.

Do your max charging current and voltage limits work?

Try running acc and cooldown without any current and voltage limits.

You may want to read about `current_workaround` (README.html).

Upgrade acc.
acc_v2021.9.11_(202109110).zip](https://github.com/VR-25/acc/files/7147378/acc_v2021.9.11_.202109110.zip)

kevin01523, 2021-09-11:
Cooldown works since 9.10.1 i just encountered a few bugs i think
As in charging is not reported or device says discharging but not really and in that situation current is fixed on the one i set and cooldown works too. I just haven't tried letting it reach the 
Stop charging capacity.just got the log while its happening and uploaded it.

cooldownCurrent=700
Do work and current do go from 1200 or 1600 into 650-699 during the 13-15 cooldown

Ill upload the log for 9.11 later I'm letting it charge for a few cycles for now

kevin01523, 2021-09-11:
I just noticed it breaks the Acc -f 
Or the charge to capacity without restrictions
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7148408/acc-logs-tulip.tar.gz)

VR-25, 2021-09-11:
What exactly is going on with `acc -f`?

kevin01523, 2021-09-11:
`
  -f|--force|--full [capacity]   Charge once to a given capacity (default: 100%), without restrictions
    e.g.,
      acc -f 95 (charge to 95%)
      acc -f (charge to 100%)
    Note: if the desired % is less than pause_capacity, use acc -e #%
`

VR-25, 2021-09-12:
[acc_v2021.9.12_(202109120).zip](https://github.com/VR-25/acc/files/7149752/acc_v2021.9.12_.202109120.zip)

kevin01523, 2021-09-12:
Only stops charging or will charge for a second then stop charging
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7149778/acc-logs-tulip.tar.gz)

VR-25, 2021-09-12:
Still, I'm unable to reproduce the issues you're facing.
Try resetting the config, rebooting, then testing `acc -d`, `acc -d` and `acc -t`.

Donzi84, 2021-09-12:
good evening,

I 've dowloaded acc_v2021.9.12_(202109120).zip, How can I install it with magisk? my magisk doen't recogniz zip file.

Thanks 
stefano

kevin01523, 2021-09-13:
Here's the log for the test i have not done anything else
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7153643/acc-logs-tulip.tar.gz)

kevin01523, 2021-09-13:
> good evening,
> 
> I 've dowloaded acc_v2021.9.12_(202109120).zip, How can I install it with magisk? my magisk doen't recogniz zip file.
> 
> Thanks
> stefano

Try flashing it via acc

VR-25, 2021-09-13:
> Here's the log for the test i have not done anything else
> [acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7153643/acc-logs-tulip.tar.gz)
> It did start charging now after i did start acc via wizard but
> Acc -d only stopped acc also charging or atleast charging status
> 
> Here's the log after changing the config and starting acc then replugged the charger
> [acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7153686/acc-logs-tulip.tar.gz)

Ok, now I found something relevant!
Working on a new build...

VR-25, 2021-09-13:
> good evening,
> 
> I 've dowloaded acc_v2021.9.12_(202109120).zip, How can I install it with magisk? my magisk doen't recogniz zip file.
> 
> Thanks
> stefano

Magisk is supposed to recognize zip files.
Are you sure that's the issue?
Anyway try using `acc -F` (wizard) or `acc -F /path/to/zip` to flash the zip.
It this also fails, you probably have a corrupt archive. Redownload the zip and try again.

VR-25, 2021-09-13:
[acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)

Donzi84, 2021-09-13:
> [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)

I've installed 9.13 but how can select it in acca?
[image]

VR-25, 2021-09-13:
You don't have to.
This almost always irrelevant.

Donzi84, 2021-09-13:
> You don't have to.
> This is almost always irrelevant.

ok, I'm testing a charching on a Samsung Tab4 sm-t335 with the acc    acc_v2021.9.13_(202109130).zip
This evening I tell you if it works.

kevin01523, 2021-09-13:
New logs for 13 i guess current control still affect the acc force charge command

Im using acca to monitor the current change without changing the config with it
[acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7156408/acc-logs-tulip.tar.gz)

Donzi84, 2021-09-13:
> [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)

It works!!!!!!! charge really stopped at 80 percent. Even after a long time the percentage no longer rises. @kevin01523 is the same for you?
@VR-25 thanks a lot

Donzi84, 2021-09-13:
> [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)

It doesn't work.  It stopped at 80% for a while but then it resumes charging. 

Stefano

VR-25, 2021-09-13:
> > [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)
> 
> It doesn't work. It stopped at 80% for a while but then it resumes charging.
> 
> Stefano

Unless you grab logs (`acc -le`) right after the occurrence of an issue, It takes longer for me to figure out what went wrong.

Donzi84, 2021-09-13:
> Unless you grab logs (`acc -le`) right after the occurrence of an issue, It takes longer for me to figure out what went wrong.

Could you please explain me how todo a log?
Thanks a lot
Stefano

VR-25, 2021-09-13:
> New logs for 13 i guess current control still affect the acc force charge command
> 
> Im using acca to monitor the current change without changing the config with it
> [acc-logs-tulip.tar.gz](https://github.com/VR-25/acc/files/7156408/acc-logs-tulip.tar.gz)
> Cooldown seems to work fine

Due to an error from the previous build, you have two charging switches set.
This may be causing issues.
Run `acc -s s=` to reset that part of the config.

VR-25, 2021-09-13:
> Could you please explain me how todo a log?
> Thanks a lot
> Stefano

Whenever to come across an issue, run `acc -le`.
It'll pack all logs into a compressed archive that you can share.
No personal information is ever included.
The commands prints the location of the generated log archive.

Donzi84, 2021-09-13:
> > Could you please explain me how todo a log?
> > Thanks a lot
> > Stefano
> 
> Whenever you come across an issue, run `acc -le`.
> It'll pack all logs into a compressed archive that you can share.
> No personal information is ever included.
> The command prints the location of the generated log archive.

where have I to write the code acc -le? in the shell? in magisk? or in acca?
thanks you in advance.
Stefano

VR-25, 2021-09-13:
Shell, as root
Two choices
1. `su` then `acc -le`
2. Just `su -c acc -le`

Donzi84, 2021-09-13:
[image]
I don't understand where save the file .gz how can I access? Thanks

VR-25, 2021-09-13:
You would have to use a root file manager.
But here's another way:
On this same terminal, _paste_ and run `cp /data/adb/vr25/acc-data/logs/acc-*.gz /sdcard/Download/`.
After this, you'll find the file in your Download directory.

Donzi84, 2021-09-13:
> You would have to use a root file manager.
> But here's another way:
> On this same terminal, _paste_ and run `cp /data/adb/vr25/acc-data/logs/acc-*.gz /sdcard/Download/`.
> After this, you'll find the file in your Download directory.

Hereby the file. Thanks a lot for your patience. :)
Stefano

[acc-logs-milletlte.tar.gz](https://github.com/VR-25/acc/files/7157282/acc-logs-milletlte.tar.gz)

VR-25, 2021-09-13:
While I analyze the logs further, see if charging control works as expected after running `acc -s s=` on terminal.

Donzi84, 2021-09-14:
> analyze the logs further, see if charging control works as expected after running `acc -s s=` on terminal.

I've tried acc -s s= nut nothing is changed. It doesn't stop at 80percent.

Thnaks 
Stefano

[26 more replies]

closed 2021-09-19 by VR-25